FreeCAD/FreeCAD #15259: BIM: Tree drag and drop access violation

maxwxyz, 2024-07-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Trying to drag an object under the Ifc Tree items results in an access violation error:
[image]

@PaddleStroke @yorikvanhavre FYI

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git)
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.3
  * Curves 0.6.40
  * dodo 1.0.1
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.21
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-07-08:
Confirmed.
This line seems to cause the problem:
https://github.com/FreeCAD/FreeCAD/blob/de508e437d13e7eb35baffceea496fb9107888cb/src/Mod/BIM/nativeifc/ifc_viewproviders.py#L322

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git)
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
```

yorikvanhavre, 2024-07-09:
Hmm it might be because of some uncaught assumptions I did there... Could you guys try the fix below?

Roy-043, 2024-07-09:
Please ignore my previous message. That may not be where the cause is.

The new code does not fix the issue I'm afraid.

Looking at the code a bit, I think the dropped object can get deleted during the drop action. That may be tricky in terms of timing.

yorikvanhavre, 2024-07-10:
New try. Works for me, but I don't have the original issue. Please test?

commit c4e55e38 (2024-07-10): BIM: fixed nativeIFC view provider - fixes #15259
PR #15341 (2024-07-10): BIM: Added delay to ifc viewprovider drag and drop
New try...
Fixes #15259
commit 41e5588e (2024-07-10): BIM: Added delay to ifc viewprovider drag and drop - fixes #15259
closed 2024-07-10 by yorikvanhavre