FreeCAD/FreeCAD #15263: Classis Stylesheet: Property view has dark background always visible

maxwxyz, 2024-07-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With overlays and transparency enabled, the property view is always dark, even if nothing is displayed in it:

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git)
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.3
  * Curves 0.6.40
  * dodo 1.0.1
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.21
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-07-07:
This is an old issue, and I cannot fix this myself with stylesheets.
The issue is that Overlay removed the background of the treeview.
And the properties panel also uses this color in not-alternative rows.
I cannot find a way to add this color again without this black box.
If I remove the black box, the cells will be click true and it looks terrible.

I think it is the color that switches to transparent when you enable transparency.
Maybe @obelisk79 and @kadet1090 can help, although Opentheme also have this issue:
[image]

kadet1090, 2024-07-07:
@maxwxyz can you reassign it to me? It requires change to the rendering side, not sure if possible but certainly impossible via stylesheets.

PR #15328 (2024-07-15): Gui: Fix background of non-alternate rows of Property View
This forces background to be rendered using `itemBackground` QProperty.

This can be done like so:
```
  Gui--PropertyEditor--PropertyEditor {
    qproperty-itemBackground: white;
  }
```

After:
[image]

Before:
[image]

Fixes: #15263

FYI: @obelisk79 @MisterMakerNL
commit ce6275e1 (2024-07-15): Gui: Fix background of non-alternate rows of Property View

This forces background to be rendered using `itemBackground` QProperty.

Fixes: #15263
closed 2024-07-15 by yorikvanhavre