FreeCAD/FreeCAD #15264: TechDraw Page Templates should be available in the Addon Manager

maxwxyz, 2024-07-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently all TechDraw page templates are included with the FreeCAD installation, so updating them for new features, adaption to new standards or corrections (typos,...) are not possible whenever you like. Currently you can manually download templates and save them in your directory.

It would be great if the Addon Manager had a section for TechDraw page / titleblock templates which could be provided by a FPA GitHub repository.

This would allow the user to browse through all available page templates, maybe sort them by locale/language, standard, page size,... and apply them directly in TechDraw.
It would also allow to update existing page templates at anytime.

@WandererFan @chennes would this be feasible?

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git)
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.3
  * Curves 0.6.40
  * dodo 1.0.1
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.21
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-07-08:
We need to at least ship some minimal set of Template with the application.   Otherwise, TD is not usable before using Addon Manager to download templates.

sliptonic, 2024-09-14:
The issue is broader than just Techdraw templates.  The same problem exists for CAM
- need to be able to install post processors, tools, vises, and other assets

Rendering
- Canned rendering templates would be nice

Materials
- we can foresee people developing and distributing custom material libraries

If Robot is further developed we could expect robot definitions to be developed and distributed this way. 

What is needed is a more generic solution to deliver assets via the addon manager.
assets are not macros and shouldn't be installed in the user macro directory.
They aren't work benches or preference packs

I believe the solution needs to address two key points:
Where should an addon (or the addon manager) install assets like this?

What development is needed in various workbenches to make them compatible with 3rd party developed assets?

adrianinsaval, 2024-09-16:
I suggest a new category of addons called "resource" that should get installed into UserResourcesDir which would be {UserDataDir}/Resources, a resource addon should generally also declare a subdirectory target inside UserResourcesDir for example for techdraw templates it can either be Templates/Techdraw in case we expect other types of templates, or Techdraw/Templates and for CAM post processors CAM/PostProcessors

adrianinsaval, 2024-09-16:
it could be benefitial to have a system resources dir and user resources dir, so we can ship templates with the system and then allow overriding them with user data, we can make a getResources function that takes a relative path such as Templates/Techdraw and optionally a file extension or more generically a file glob expression and returns a list of the available files from system + user resources, for any resource name that is available on both the user resources will take precedence over the system ones