FreeCAD/FreeCAD #15265: Materials should be available in the Addon Manager

maxwxyz, 2024-07-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It would be great if the new materials would be available in the Addon Manager, maybe add sections and filters to browse by application, type, properties, hatching,...
This way a user could download all needed materials from a reliable source and the materials could also be updated via the Addon Manager anytime.

@davesrocketshop I know you mentioned somewhere that most sites do not allow the distribution of material information from their databases but maybe it could be an open source / user defined repository for the FreeCAD materials. We currently distribute materials with the FreeCAD installation as well.

@chennes FYI

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git)
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.3
  * Curves 0.6.40
  * dodo 1.0.1
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.21
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

davesrocketshop, 2024-07-07:
Discussed this with @chennes a while ago. This is not an addon manager feature, but a Materials add on. There's no reason why this can't be done, but some core materials still need to be in the main product, such as a "default" material and many of the existing appearances. There is already an ability for a module to define both models and materials.

The same problem applies in that it can't include copyright material.

There are plans for post-1.0 features such as database and web api features that may make this unnecessary, or at least work together to solve the same problem.

MisterMakerNL, 2024-07-20:
Agree with this, any info that is external would be best in add-ons, like materials drawing standards. Part libraries.