FreeCAD/FreeCAD #15267: Opening Assembly WB creates new toolbar with a single icon

hasecilu, 2024-07-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Opening/Reloading Assembly WB creates those toolbars with that button that appears to do nothing.

Also appeared when using "Insert component" command and **even swapping from workspace on Hyprland** (Wayland prblem?) creates another one. All commands can be moved to same toolbar but not funny moving more than 20 icons.

[image]

No MenuTExt, the tooltip of a command was: 
[image]

### Full version info

```shell
OS: Arch Linux (Hyprland/hyprland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * ThreadProfile 1.90.0
  * FreeCAD-Movie 2024.2.15
  * freecad.gears 1.2.0
  * PieMenu 1.7.3
  * freecad_metal_workbench 0.0.1
  * Ship 2022.4.11
  * OpenTheme 2024.7.3
  * FreeGrid 2.0.0
  * FreeCAD-Reinforcement
  * Defeaturing 1.2.2
  * FreeCAD_SheetMetal 0.4.16
  * FreeCAD-Pyramids-and-Polyhedrons
  * yaml-workspace 0.1.2
  * woodworking 0.21.2.33771
  * Assembly4 0.50.10
  * FastenersWB 0.5.24
  * lattice2 1.0.0
  * Rocket 3.3.0
  * dodo 1.0.1
  * FreeCAD-motion
  * cadquery_module
  * Profiles 0.0.1
  * Alternate_OpenSCAD 1.0.0
  * osh-autodoc-workbench 0.2.3
  * CfdOF 1.27.4
  * A2plus 0.4.64a
  * FreeCAD_WeldfFeature 0.1.0
  * kicadStepUpMod 11.2.4
  * Nodes 0.1.36
  * Marz 0.1.5
  * frame 0.1.1
  * pcb 6.2023.1
  * FreeCAD_assembly3 0.12.2
  * Render 2024.7.4
```

### Subproject(s) affected?

Assembly

### Anything else?

Not sure if was triggered by opening an Assembly 4 project, but FreeCAD was restarted various times.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-07:
I cannot reproduce on
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git)
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.3
  * Curves 0.6.40
  * dodo 1.0.1
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.21
```

hasecilu, 2024-07-07:
I filtered all external WB and the icon that appears is from [WeldFeature](https://github.com/alexneufeld/FreeCAD_WeldfFeature/blob/main/freecad/weldfeature/resources/WeldFeature.svg) WB.
Probably tried to add command for testing purposes.
Is normal for `__main__.toolbar_manipulation` be triggered even when no interection with toolbars is done?
```python
def toolbar_manipulation(name):
    import FreeCAD

    FreeCAD.Console.PrintLog(f"workbench name: {name}\n")
    if name == "AssemblyWorkbench":
        from freecad.weldfeature import ICONPATH
        import os
        import FreeCADGui

        mw = FreeCADGui.getMainWindow()
        from PySide import QtWidgets, QtGui

        my_tool_button = QtWidgets.QToolButton()
        my_tool_button.setIcon(QtGui.QIcon(os.path.join(ICONPATH, "WeldFeature.svg")))
        my_tool_button.clicked.connect(lambda: FreeCADGui.runCommand("WeldFeature_Add"))
        my_tool_bar = QtWidgets.QToolBar("Assembly Features")
        my_tool_bar.addWidget(my_tool_button)
        mw.addToolBar(my_tool_bar)

# The following 2 lines are important because InitGui.py files are passed to the exec() function...
# ...and the runMacro wouldn't be visible outside. So explicitly add it to __main__
__main__.toolbar_manipulation = toolbar_manipulation

# Connect the function that runs the macro to the appropriate signal
FreeCADGui.getMainWindow().workbenchActivated.connect(toolbar_manipulation)

# Add the GUI command
FreeCADGui.addCommand("WeldFeature_Add", AddWeldFeatureCommand())
```

maxwxyz, 2024-07-07:
not sure but you should add an issue in the WeldFeature Repo then. Could this here be closed then?

hasecilu, 2024-07-07:
Yes, I'm into that, sorry for false alarm.

closed 2024-07-07 by hasecilu