FreeCAD/FreeCAD #15268: Classic theme: Sketcher dialog not legible

maxwxyz, 2024-07-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Classic stylesheet with overlays "auto" makes the sketcher task panel not legible.
It is white text on white background:

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git)
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.3
  * Curves 0.6.40
  * dodo 1.0.1
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.21
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-07:
It's also white in almost any task dialog: 
[image]

MisterMakerNL, 2024-07-07:
I fixed some things in these woods of the stylesheet so I am pretty sure it is fixed here:
https://github.com/FreeCAD/FreeCAD/pull/14749
I did tested it and did not see this happening.

furgo16, 2024-07-08:
I can confirm on

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38039 (Git) Snap 1067
Build type: Release
Branch: main
Hash: 7a9fd501c3ea0e259f843dc21f488ef6af43c201
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.7.3
```

maxwxyz, 2024-07-10:
@MisterMakerNL does the fix also fix dropdown menus in task dialogs with classic:
[image]

maxwxyz, 2024-07-10:
Also scrolling in a text field renders the text white: 
[image]

closed 2024-07-15 by chennes