FreeCAD/FreeCAD #15270: Active container and active feature colors overlay conflict

MisterMakerNL, 2024-07-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Overlay in classic uses white text in transparant mode, because the background is dark.
This white text conflicts with the default active container colors.
Since these colors aren't stylesheet controlled they cannot change with the state of the overlay.
I have no idea why these colors aren't just controlled by the stylesheet.
But we ether remove them from the menu and move them too a stylesheet, or we change them to a color that works fine with white and black text.
[image]

### Full version info

```shell
NA
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-08:
@MisterMakerNL this is solved in the latest weekly?

closed 2024-08-08 by maxwxyz

MisterMakerNL, 2024-08-08:
Not sure why but it's solved in my buikd.