FreeCAD/FreeCAD #15275: Tool shapes not copied by "Setup ToolBit working directory"

spanner888, 2024-07-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
Downloaded above FC
Deleted ~/.local/share/FreeCAD/
Deleted ~/.config/FreeCAD/
Ran FC
Opened CAM WB
Created new doc
Opened report view
Selected: Menu CAM - ToolBitLibrary Editor
Click Yes to Setup ToolBit working directory
Created new folder "Tools" & "Chose" it
Clicked yes THREE times to copy dailogs.
Library editor opens.

Output from console tree command shown below shows NO shape files copied. This means user cannot easily create not ToolBits, or edit default Tool Shapes.

`tree ~/.local/share/FreeCAD/
/home/spanner888/.local/share/FreeCAD/
└── Macro
    └── Tools
        ├── Bit
        │   ├── 45degree_chamfer.fctb
        │   ├── 5mm_Drill.fctb
        │   ├── 5mm_Endmill.fctb
        │   ├── 5mm-thread-cutter.fctb
        │   ├── 60degree_Vbit.fctb
        │   ├── 6mm_Ball_End.fctb
        │   ├── 6mm_Bullnose.fctb
        │   ├── probe.fctb
        │   └── slittingsaw.fctb
        ├── Library
        │   └── Default.fctl
        ├── README.md
        └── Shape`

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
```

### Subproject(s) affected?

CAM/Path

### Anything else?

There are other CAM issues about ToolBits/shapes, but could not find any mentioning missing Shape files after running "Setup ToolBit working directory"

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-08:
@spanner888 does this work in the stable 0.21.2?

spanner888, 2024-07-14:
> @spanner888 does this work in the stable 0.21.2?

No, exact same issue, Shapes folder is empty.

Followed steps reported initially, only changes due to Path-CAM migration.

spanner888, 2024-08-10:
Looked at the code and see that Shape dir is explictly NOT copied during the initial Library setup.

https://github.com/FreeCAD/FreeCAD/blob/aab0b31aec8d30691c3b012023e7a26fcc627b2a/src/Mod/CAM/Path/Tool/Gui/BitLibrary.py#L140

Can also see in much older code the Tool Shape files were also not copied.

But below & possibly elsewhere always gave me the impression that shape files were copied.
[image]
Copies of the default toolshapes, toolbits and library are copied to this directory.

Can anyone clarify whether the documentation or the code is the item that should be updated?

@Russ4262 @sliptonic

Russ4262, 2024-08-11:
Morning.
Using the 'Steps to reproduce' on my machine also fails to populate the newly created 'Shape' folder.  My initial opinion is to remove the 'if dir != "Shape":' line of code and shift the contents of that clause left one indent level so that it is run for each iteration of the 'for dir in subdirlist:' parent loop.  Perhaps other devs know the reason to purposefully omit population of the 'Shape' folder, especially since the user will be prompted for permission before doing so.

Best Regards,
Russell

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * FC_SU
  * freecad.gears 1.0.0
  * freecad.pathred
  * Z_MacroStartup
```

sliptonic, 2024-08-17:
I think the current behavior is correct.  The Shape directory is created to hold any custom shapes created by the user.  The  shapes provided by FreeCAD shouldn't be copied. 
Is there a rationale for wanting to copy the shapes that are shipped with FreeCAD?

sliptonic, 2024-08-17:
When creating new toolbits, it should show toolshapes from the user location and from the builtin collection.  This works for me on linux.  

When we implemented this originally, we didn't know how often new shapes would be created.  In practice, it's pretty rare.  Editing the builtin shapes seems particularly dangerous as an end-user activity and should be discouraged.

spanner888, 2024-08-17:
I raised issue as I 'believed' shapes were supposed to be copied, possibly because the wiki says they are as noted above.

I agree editing system shapes is a bad idea, also that user copies of system shapes can also cause issues.

Hence my question above "Can anyone clarify whether the documentation or the code is the item that should be updated?"

I can update the wiki if that is required.

spanner888, 2024-09-02:
Wiki updated, closing issue

closed 2024-09-02 by spanner888