FreeCAD/FreeCAD #15277: In the Preferences dialog, the "Preference pack" section has several usability issues

raindropsfromsky, 2024-07-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the Preferences dialog, the "Preference pack" section has several usability issues:
1. The "apply" text is clipped at bottom (the height of the button is not enough)
    [image]
2. When we click on a button, it does not change color to show "applied" state
3. The user gets no clue how the **"Apply"** button is supposed to work:
     - Like a radio button? (only one of them can be selected at a time), or
     - Like a checkbox? (any number of buttons can be selected at a time).
4. There is no apparent way to "un-apply" a pack!
    - Does it toggle on/off if clicked multiple times?
5. It provides a list of themes. But there is already a **Theme** drop-down in the **Preferences** dialog.
   [image]
   As a result, it is unclear which theme would be effective at any point of time!
6. The **Themes** drop-down list and the **Preference pack** provide the same functionality, and yet they are located far apart. 

### Full version info

```shell
OS: Windows 11 build 22631
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.38021 (Git)
Build type: Release
Branch: flavor
Hash: a34043c4fa4efa3c871e376a4e1536dcdeb86ebc
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/India (en_IN)
Installed mods: 
  * 3DfindIT 1.2.0
  * Behave-Dark-Colors 0.1.1
  * fasteners 0.5.24
  * Freecad-Built-in-themes-beta 1.2.2
  * FreeCAD-themes 2024.7.24
  * Ondsel-Lens 2024.3.26.01
  * OpenDark 2024.3.13
  * OpenTheme 2024.7.3
  * PieMenu 1.7.0
  * ProDarkThemePreferencePack 1.0.0
  * SearchBar 1.0.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-07-08:
Related issue: https://github.com/FreeCAD/FreeCAD/issues/13714

furgo16, 2024-07-09:
I believe points 5., 6., and 7. are already addressed in the development version by the change that hides Preference Packs from the preference pack manager if they are themes. An alternative solution might have been to disable the <kbd>Apply</kbd> button for themes there.

raindropsfromsky, 2024-07-09:
Well, TBH this dialog needs a major revamp, this is the most logical place to launch Preference Pack Editor.
So it needs an EDIT button for each pack.

Note that the Addon Manager does not edit the Preference Packs, nor does it save a local copy.
IINW, this facility is not there in FreeCAD at all.

If this is out of scope for this issue, I can elaborate this as a separate feature request.

raindropsfromsky, 2024-07-09:
IINW, a Preference Pack may or may not include a theme in its definition.
But if there is a theme included in a Preference Pack, it will always have a few additional parameters defined.

That is the reason the UI has a separated Preference Packs from "pure" themes, right?

obelisk79, 2024-07-18:
Preference packs are a versatile system which allows for comprehensive theming, which is why there is a seeming overlap. During the 1.0 development cyclem the ability to flag a preference pack as a theme was added. Older themes may not have been properly updated so they still show in the preference pack list.

Alternatively, a preference pack can exist which isn't a theme in any way at all. As an example, I intend to include an OpenTheme preference pack which arranges toolbars and panels differently than the existing default. It will also separate several preferences currently applied by OpenThemes away from the themes themselves.

There is certainly some confusion involved with all of this, but it isn't so bad. The problem really stems from preference packs which aren't properly flagged  as themes.

luzpaz, 2024-07-24:
Should this ticket be split ? If not, please propose a way to keep track of what is valid and invalid + complete or incomplete ?

obelisk79, 2024-07-24:
1. Is absolutely valid
.
2. 3. And 4. Are unactionable without reconsideration of how preference packs are applied. I don't know if this is something @chennes is willing to consider. 
.
5. 6. And 7 are merely a misunderstanding of the distinction between what a theme and preference pack each are.

krushia, 2024-08-18:
Something I've noticed:

1. Have FreeCAD Classic Theme enabled, but with some custom color tweaks
2. Save a new preference pack
3. Switch to FreeCAD Dark
4. Apply
5. Apply preference pack saved before

Settings will apply, but Preferences will show as FreeCAD Dark. Switching that to Classic (which is already applied but with some extra user settings) will wipe out color settings. This is confusing.

Might also relate to #15890

obelisk79, 2024-08-18:
Hi Ken, thanks for your comment. This is a separate issue and I recommend you open a new ticket. My assumption here, is that saving a new preference pack doesn't record the selection of the theme pulldown. Possibly some oversight, or just an unintended consequence of some other decision. @chennes is the most knowledgeable about preference packs and is likely the best person to address the concern.

obelisk79, 2024-08-18:
@luzpaz @maxwxyz I didn't think this issue is relevant anymore after all of the recent changes to themes. The single valid issue raised in the initial ticket is resolved.

It may be time to close this ticket.

closed 2024-08-18 by maxwxyz

chennes, 2024-08-18:
It is really that the theme pulldown is populated by preference packs that are marked as a "theme", and that is not getting set in a saved pack. So the menu needs some kind of "Custom" entry, but the logic is going to get confusing and ugly.

raindropsfromsky, 2024-08-27:
> @luzpaz @maxwxyz I don't think this issue is relevant anymore after all of the recent changes to themes. The single valid issue raised in the initial ticket is resolved.
> 
> It may be time to close this ticket.

Well, I have this additional observation (related to issues 2-3-4):
While the **Preference packs** section alloows us only to apply the pack, we cannot remove them from this place.
In fact, the removal button is provided in the **Addon Manager**, which is two windows away from the expected location! 

Most logical placement:
#1: Right in the Reference packs section (1)
#2: In the Manage preference packs dialog box. (2)
[image]

IMHO issues 2-3-4 are legitimate isses.
We can call them "enhancement" or "feature request".
Nevertheless, they should be addressed.
If necessary, I can raise new ticket(s) for these issues.

obelisk79, 2024-08-27:
Why would removal place be separate from the installation place? I think you make a reasonable point, but the current arrangement isn't illogical either.

raindropsfromsky, 2024-08-27:
Let me clarify issues 5-7 by rephrasing those points:

As the screenshot in my first post shows, the UI allows _simultaneous_ application of mismatching theme and preference pack!
Where a preference pack may contain a theme, and then additional parameters.
(This is my point-7)

Of course, only one of them can take effect.
That means the UI is not designed to prevent conflict.

If a preference pack contains a theme, then the independent **Theme** field should be grayed out.
(This is my point-5)

Further, this linkage becomes clear only if these two controls are laid next to each other in the UI.
(This is my point-6).

raindropsfromsky, 2024-08-27:
> Why would removal place be separate from the installation place? I think you make a reasonable point, but the current arrangement isn't illogical either.

Fair enough! The Addon Manager must have the provision for removal. 
I was not against that, in fact.