FreeCAD/FreeCAD #15278: When I apply a new theme, the dialog grays out and becomes unresponsive indefinitely, and gives "cannot access classic light theme" error.

raindropsfromsky, 2024-07-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I apply a new theme, the dialog grays out and becomes unresponsive indefinitely. 
I had to kill Ondsel to exit. 

This happens repeatedly.

I get multiple lines of this error:
```Cannot access C:\Users\naray\AppData\Roaming\Ondsel\Mod\Freecad-Built-in-themes-beta\Classic light```

Note that this error pops up despite the fact that neither my current theme nor my newly selected theme are "classic light"!

I am running the latest nightly Ondsel on Windows 11.
My laptop has Intel i3 8310U CPU and UHD 620.
12 GB DDR4 RAM

When I face the issue, the system is not overloaded:
RAM load: 36%
CPU load: 30%.
GPU load: 1%.
All the other apps continue to work normally. Only Ondsel becomes unresponsive.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.38021 (Git)
Build type: Release
Branch: flavor
Hash: a34043c4fa4efa3c871e376a4e1536dcdeb86ebc
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/India (en_IN)
Installed mods: 
  * 3DfindIT 1.2.0
  * Behave-Dark-Colors 0.1.1
  * fasteners 0.5.24
  * Freecad-Built-in-themes-beta 1.2.2
  * FreeCAD-themes 2024.7.24
  * Ondsel-Lens 2024.3.26.01
  * OpenDark 2024.3.13
  * OpenTheme 2024.7.3
  * PieMenu 1.7.0
  * ProDarkThemePreferencePack 1.0.0
  * SearchBar 1.0.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-07-08:
This might have been fixed in the weekly builds: https://github.com/FreeCAD/FreeCAD/pull/14820

maxwxyz, 2024-07-08:
@raindropsfromsky please test in the latest weekly. On my system it tkes up to 2 minutes but then it works

raindropsfromsky, 2024-07-09:
> On my system it tkes up to 2 minutes but then it works

I am testing on the latest weekly of Ondsel (which may be delayed by a week compared to FreeCAD).
I hope that Ondsel will catch up next week.

But even after that, the best I can hope is 2 minutes delay.
Even that much delay is unreasonable, isn't it?

furgo16, 2024-07-09:
The delay might be a platform-specific issue. On my Linux system it's a matter of seconds to switch themes.

raindropsfromsky, 2024-07-09:
Well, the earlier versions used to work fast on the same system.
This delay is happening only in recent versions.

luzpaz, 2024-07-11:
Should this be moved to the Ondsel issue queue ?

MisterMakerNL, 2024-07-20:
Should be fixed in latest weekly!!

closed 2024-07-21 by maxwxyz