FreeCAD/FreeCAD #15281: The Addon Manager provides Preference packs, but not themes. There is no facility to filter for themes, either.

raindropsfromsky, 2024-07-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Addon Manager provides Preference packs, but not themes.
Even the filter in Addon Manager shows Preference packs, but not themes.
[image]

Thus, the themes seem to be completely deprecated in the Addon Manager.
This is confusing for a user.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of Ondsel: 64-bit
Version: 2024.2.0dev.38021 (Git)
Build type: Release
Branch: flavor
Hash: a34043c4fa4efa3c871e376a4e1536dcdeb86ebc
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/India (en_IN)
Installed mods: 
  * 3DfindIT 1.2.0
  * Behave-Dark-Colors 0.1.1
  * fasteners 0.5.24
  * Freecad-Built-in-themes-beta 1.2.2
  * FreeCAD-themes 2024.7.24
  * Ondsel-Lens 2024.3.26.01
  * OpenDark 2024.3.13
  * OpenTheme 2024.7.3
  * PieMenu 1.7.0
  * ProDarkThemePreferencePack 1.0.0
  * SearchBar 1.0.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-08:
Themes are the preference packs as I understand it.
@obelisk79 correct?

closed 2024-07-08 by maxwxyz

obelisk79, 2024-07-08:
There's some validity to the original issue. That is one of discoverablity. Themes are preference packs. But preference packs are not necessarily themes as they are more versatile. However, all existing preference packs *are* themes. I think the discoverablity aspect is somewhat covered by the recently added get more link next to the themes pulldown in preferences. So I would keep this closed as not planned for now.

raindropsfromsky, 2024-07-09:
> Themes are the preference packs as I understand it.

Well, I know that this is work in progress, but IMHO the User Manual Wiki should clarify the design philosophy.

I hope that we do not have to deal with four different items:
1. Standard themes
2. Locally modified themes (saved by user with a different name)
3. Standard Preference Packs
4. Locally modified Preference Packs (saved by user with a different name)

It is best to ditch the first two varieties, and simplify the UI.

obelisk79, 2024-07-09:
preference packs which are themes will now populate the themes dropdown, instead of showing in a preference pack list as exists in 0.21

Also, I strongly suggest you uninstall the dedicated OpenDark preference pack. It is no longer maintained. I've moved all OpenDark/OpenLight development to the OpenTheme repo.

[image]

raindropsfromsky, 2024-07-09:
Thanks for the heads up!
Kindly have a look at Issue #15277 also. 

Thanks in advance!

raindropsfromsky, 2024-07-09:
BTW, why separate "pure" themes and "preference packs" (which are essentially modified themes)?
Why not have everything in one place?

obelisk79, 2024-07-09:
Preference packs can be any setting saved in your FreeCAD config file and not just theme information. So things like panels being enabled, or panel position, or hiding/disabling of workbenches etc. there is a lot of similarity but preference packs can exist that are not themes at all.