FreeCAD/FreeCAD #15284: [1.0 Bughunt] CAM ToolBit Library NOT always saved

spanner888, 2024-07-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Select: Menu CAM - ToolBitLibrary Editor
Create New, empty Library
Add existing tool
Click on any other existing library - eg Default, or just on the Window Close x.
Click on Library just created and it will be empty.
If after "Add existing tool", user closes the library window with close button, tool is saved.
Also occurs if multiple tools added.

In this case a warning SHOULD be given to user, but no message is given.

Possibly some relation with #14909, as that is also about saving ToolBits. but detail is quite different.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of Ondsel: 64-bit
Version: 2024.2.1.37226 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 2024.2.1)
Hash: 1a1bd7290494aef925a026c87b808ac58d53c3b0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Installed mods:
  * Ondsel-Lens 2024.7.5.02

OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-08:
Does this work in the stable 0.21.2 version?

spanner888, 2024-07-14:
> Does this work in the stable 0.21.2 version?

No. Same issue, where added tool is only saved via close, any other action does not save added tool/s.

spanner888, 2024-08-10:
Looked at the code and see that Shape dir is explictly NOT copied during the initial Library setup in BitLibrary.py

Can also see in much older code the Tool Shape files were also not copied.

But below & possibly elsewhere always gave me the impression that shape files were copied.
[Wiki-  CAM Tools - Organisation](https://wiki.freecad.org/CAM_Tools#Organization)

> Copies of the default toolshapes, toolbits and library are copied to this directory.

I would prefer that the Shapes are also copied. I believe there have been queries in the forum, but cannot find, not even in recent queries about creating new shape files.

Can anyone indicate if this is a documentation issue leading to user/my misunderstanding or should code change to match the documentation?
? @sliptonic @Russ4262

luzpaz, 2024-09-01:
bumping for visibility

closed 2024-09-10 by sliptonic