FreeCAD/FreeCAD #15285: module 'ifcopenshell.util' has no attribute 'element'

Mannshoch, 2024-07-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. snap install freecad --beta
2. Download: https://openifcmodel.cs.auckland.ac.nz/
3. Open the IFC
4. 
```
08:30:50  Migrating Start Workbench to Start command... done.
08:30:57  propsReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.179\" (uid=1000 pid=13776 comm=\"/snap/freecad/1069/usr/bin/FreeCAD - --single-inst\" label=\"snap.freecad.freecad (enforce)\") interface=\"org.freedesktop.DBus.Properties\" member=\"GetAll\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=1678 comm=\"/usr/sbin/NetworkManager --no-daemon\" label=\"unconfined\")"
08:30:57  nmReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.179\" (uid=1000 pid=13776 comm=\"/snap/freecad/1069/usr/bin/FreeCAD - --single-inst\" label=\"snap.freecad.freecad (enforce)\") interface=\"org.freedesktop.NetworkManager\" member=\"GetDevices\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=1678 comm=\"/usr/sbin/NetworkManager --no-daemon\" label=\"unconfined\")"
08:30:57  "Object path cannot be empty"
08:30:57  "Empty widget item in QVBoxLayout 'verticalLayout'."
08:31:14  Invalid version specified for tag version in Addon Trails: 2022-01
08:31:43  pyException: Traceback (most recent call last):
  File "<string>", line 5, in <module>
  File "/snap/freecad/1069/usr/Mod/BIM/nativeifc/ifc_import.py", line 52, in open
    insert(filename, doc.Name, singledoc=True)
  File "/snap/freecad/1069/usr/Mod/BIM/nativeifc/ifc_import.py", line 83, in insert
    prj_obj = ifc_tools.convert_document(document, filename, shapemode, strategy)
  File "/snap/freecad/1069/usr/Mod/BIM/nativeifc/ifc_tools.py", line 126, in convert_document
    create_children(document, ifcfile, recursive=False)
  File "/snap/freecad/1069/usr/Mod/BIM/nativeifc/ifc_tools.py", line 324, in create_children
    result.extend(create_child(obj, child))
  File "/snap/freecad/1069/usr/Mod/BIM/nativeifc/ifc_tools.py", line 291, in create_child
    child = create_object(element, doc, ifcfile, mode)
  File "/snap/freecad/1069/usr/Mod/BIM/nativeifc/ifc_tools.py", line 258, in create_object
    ifc_layers.add_layers(obj, ifcentity, ifcfile)
  File "/snap/freecad/1069/usr/Mod/BIM/nativeifc/ifc_layers.py", line 95, in add_layers
    layers = ifcopenshell.util.element.get_layers(ifcfile, element)
<class 'AttributeError'>: module 'ifcopenshell.util' has no attribute 'element'
```
5. Activate Bim Workbench
6. Open IFC file
7. Same error message appear if I use Native  import.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38050 (Git) Snap 1069
Build type: Release
Branch: main
Hash: 4d90e1655a9756c1075993f77b0d760da7404253
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-07-08:
> 2. Download: https://openifcmodel.cs.auckland.ac.nz/

Which specific file did you download from there?

For what is worth, I downloaded the `TestIFC` file (that website does not seem to provide direct links) and I'm getting the same error:

```python-traceback
10:37:08  pyException: Traceback (most recent call last):
  File "<string>", line 5, in <module>
  File "/snap/freecad/1070/usr/Mod/BIM/nativeifc/ifc_import.py", line 52, in open
    insert(filename, doc.Name, singledoc=True)
  File "/snap/freecad/1070/usr/Mod/BIM/nativeifc/ifc_import.py", line 83, in insert
    prj_obj = ifc_tools.convert_document(document, filename, shapemode, strategy)
  File "/snap/freecad/1070/usr/Mod/BIM/nativeifc/ifc_tools.py", line 126, in convert_document
    create_children(document, ifcfile, recursive=False)
  File "/snap/freecad/1070/usr/Mod/BIM/nativeifc/ifc_tools.py", line 324, in create_children
    result.extend(create_child(obj, child))
  File "/snap/freecad/1070/usr/Mod/BIM/nativeifc/ifc_tools.py", line 291, in create_child
    child = create_object(element, doc, ifcfile, mode)
  File "/snap/freecad/1070/usr/Mod/BIM/nativeifc/ifc_tools.py", line 258, in create_object
    ifc_layers.add_layers(obj, ifcentity, ifcfile)
  File "/snap/freecad/1070/usr/Mod/BIM/nativeifc/ifc_layers.py", line 95, in add_layers
    layers = ifcopenshell.util.element.get_layers(ifcfile, element)
<class 'AttributeError'>: module 'ifcopenshell.util' has no attribute 'element'
```

Using the snap weekly build as well:

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38051 (Git) Snap 1070
Build type: Release
Branch: main
Hash: dfe0c5d4ee4841f11e62da1104e322e95834aba2
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.7.3
  * Curves 0.6.40
  * ezydxf
  * Ondsel-Lens 2024.7.5.02
  * Assembly4 0.50.15
```

yorikvanhavre, 2024-07-08:
No problem for me:

[image]

But:

> <class 'AttributeError'>: module 'ifcopenshell.util' has no attribute 'element'

This makes me think: Too old IfcOpenShell version. How did you guys get IfcOpenShell Bundled with FreeCAD?

furgo16, 2024-07-08:
Ah, it seems at least I don't have the latest ifcopenshell installed:

```
freecad.pip index versions ifcopenshell
ifcopenshell (0.7.0.240627)
Available versions: 0.7.0.240627, 0.7.0.240521, 0.7.0.240518, 0.7.0.240423, 0.7.0.240418, 0.7.0.240406
  INSTALLED: 0.7.0.231218
  LATEST:    0.7.0.240627
```

furgo16, 2024-07-08:
Looking at the raw logs of the [latest build](https://github.com/FreeCAD/FreeCAD-snap/actions/runs/9831654576/job/27139346006) seems to show the correct ifcopenshell version is being picked up:

```
2024-07-08T01:57:16.8259321Z :: Collecting ifcopenshell
2024-07-08T01:57:16.8764677Z ::   Downloading ifcopenshell-0.7.0.240627-py310-none-manylinux_2_31_x86_64.whl (51.7 MB)
```

yorikvanhavre, 2024-07-08:
I don't know much about snaps :sweat_smile: 
That would be excellent if it works! One more problem out of the way :)

furgo16, 2024-07-08:
> I don't know much about snaps 😅

Well, I'm just an expert at... poking in the dark :-)

Unfortunately, I can still reproduce the bug, having confirmed that I now have the latest IfcOpenShell installed as available from PyPI.

From the CLI (package version):

```
freecad.pip index versions ifcopenshell
WARNING: pip index is currently an experimental command. It may be removed/changed in a future release without prior warning.
ifcopenshell (0.7.0.240627)
Available versions: 0.7.0.240627, 0.7.0.240521, 0.7.0.240518, 0.7.0.240423, 0.7.0.240418, 0.7.0.240406
  INSTALLED: 0.7.0.240627
  LATEST:    0.7.0.240627
```

On FreeCAD's Python console (software version, does the version string correspond to the [git commit SHA](https://github.com/IfcOpenShell/IfcOpenShell/commit/c7830e9e5)? ):

```python
import ifcopenshell
>>> ifcopenshell.version
'v0.7.0-c7830e9e5'
```

furgo16, 2024-07-08:
I installed the latest weekly AppImage and it seems to load the file just fine. The ifcopenshell reported is:

```
>>> import ifcopenshell
>>> ifcopenshell.version
'0.7.0'
```

So it seems that both the snap and AppImage packages use the same major.minor.patch version of IfcOpenShell. This seems to disprove the "too old IfcOpenShell theory", but I'm not an expert on IfcOpenShell, so I might be missing something.

yorikvanhavre, 2024-07-08:
All these versions are 0.7.0... Unfortunately some have fixes that some others don't, and it's super hard to know which one is correct

furgo16, 2024-07-08:
But I think the issue is that `pip` is ubiquitous enough for people who are not on AppImage to use it and potentially hit that bug.

furgo16, 2024-07-08:
On snap, it might be possible to pin to the same version the AppImage uses, if it's still available in pip

yorikvanhavre, 2024-07-08:
I compile IfcOpenShell myself and it works fine here, so I guess the latest version is always the good one

furgo16, 2024-07-08:
I've just tested it again with the .deb package from the daily PPA, using the latest pip IfcOpenShell package. Same Python traceback as the original report. I've ran out of ideas now.

furgo16, 2024-07-08:
Also reported at https://forum.freecad.org/viewtopic.php?p=767720

furgo16, 2024-07-08:
Seeing that the reported failure is at:

```python-traceback
  File "/snap/freecad/1070/usr/Mod/BIM/nativeifc/ifc_layers.py", line 95, in add_layers
    layers = ifcopenshell.util.element.get_layers(ifcfile, element)
<class 'AttributeError'>: module 'ifcopenshell.util' has no attribute 'element'
```

I tried the following on the snap package:

1. Try to invoke an `element` module function directly => FAIL (`element` not shown in autocompletion; it is shown in the AppImage though)
   [image]
1. Try to import the `element` module => PASS (which is now odd)
   [image]
1. Try to invoke an `element` module function from the `import as` => PASS
   [image]

Another difference that I found on the AppImage, other than the fact that it passes step 1, is that `import ifcopenshell` takes 5-10 s, whereas on the snap package it takes 30s if not longer.

Roy-043, 2024-07-08:
Maybe @Moult can help us out.

Moult, 2024-07-09:
The reason is that you haven't actually imported the module. Add `import ifcopenshell.util.element` to the file and you should be good.

Right now the import statement isn't there. Therefore it by default won't work unless some previous module has executed and imported it there (or indirectly).

In older versions of IfcOpenShell, there was some namespace creep which meant that import ifcopenshell would indirectly also import some other things (possibly ifcopenshell.util.element). This has been cleaned up and now whenever you want to use a submodule, you need to import it explicitly. e.g. `import ifcopenshell.util.element` (or if you use the API, things like `import ifcopenshell.api.root` etc.

yorikvanhavre, 2024-07-09:
Oh indeed it's me who was running an outdated IfcOpenShell version :sweat_smile: 
Thanks for the debugging guys!

Mannshoch, 2024-07-09:
May allow me a question here a bit offtopic. I'm not sure if it is intended that way, that if BIM workbench is not active that the normal Import happen? 
I assumed that If you do import an IFC but does not have the BIM Workbench Active, It will be either loaded or at least the New IFC import will anyway automatically used. (I assumed the original IFC import got replaced.)

yorikvanhavre, 2024-07-09:
> I assumed that If you do import an IFC but does not have the BIM Workbench Active, It will be either loaded or at least the New IFC import will anyway automatically used. (I assumed the original IFC import got replaced.)

Yes, that's the case indeed. (The old importer is still there and you can still use it via python - https://wiki.freecad.org/Arch_IFC )

PR #15317 (2024-07-10): BIM: Fixed IfcOpenShell imports
Fixes #15285
commit 855bdeb3 (2024-07-10): BIM: Fixed IfcOpenShell imports - fixes #15285
closed 2024-07-10 by yorikvanhavre