FreeCAD/FreeCAD #15286: SpaceMouse Compact not recognised on macOS Sonoma with 3DxWare 10.8.4

tamasgal, 2024-07-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a SpaceMouse Compact with the latest 3DxWare 10.8.4 driver on macOS Sonoma 14.3 but the mouse does not work at all in FreeCAD 0.21.2. I can see the SpaceMouse settings in the `Customize` window but `Calibrate` does nothing and I also can't add any button actions, nor do I see the exact model of my mouse in the list (SpaceMouse Compact).

I have found a few related issues, e.g. https://github.com/FreeCAD/FreeCAD/issues/14508 which uses a slightly older patch version of 3DxWare (10.8.2) and apparently there is some functionality of the mouse, but I can't get mine working at all.

Is there any other configuration needed?

This is what I see in FreeCAD:

[image]
[image]

### Full version info

```shell
[code]
OS: macOS 14.3
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-08:
@tamasgal could you please try the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
@pskowronskiTDx FYI

tamasgal, 2024-07-08:
Works like a charm, after activating "Flip Y/Z" :)

closed 2024-07-09 by maxwxyz