FreeCAD/FreeCAD #15296: The menu is missing in Chinese

HaydenHu, 2024-07-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
[image]
After setting the default English language to Chinese, the menu text is blocked, and the problem is still the same

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git)
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Chinese/China (zh_CN)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-07-08:
I have happen this too me too, but weird thing is it when away after rebooting my pc.
So I have no idea what happend.

maxwxyz, 2024-07-08:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/14920

closed 2024-07-08 by maxwxyz