FreeCAD/FreeCAD #15302: Expressions _math.pi and _math.e are broken

krushia, 2024-07-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD provides limited access to the Python math module via expressions using the pseudoproperty `_math`. `_math.pi` and `_math.e` don't seem to work, presumably because of the builtin `pi` and `e` constants (which are hardcoded and not related to the Python math ones at all) confusing the expression parser.

For correctly working examples, try any of these in an expression:
```
_math.inf
_math.nan
_math.tau
```

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.40
  * A2plus 0.4.65
  * DynamicData 2.62.0
  * Assembly4 0.50.14
  * dodo 1.0.1
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9
  * Design456 0.0.1
  * fasteners 0.5.24
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * 3D_Printing_Tools
  * Pyramids-and-Polyhedrons
  * workfeature (Disabled)
  * MnesarcoUtils 0.2.5
  * OpenTheme 2024.7.3
  * Manipulator 1.5.7
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * ThreadProfile 1.90.0
  * freecad.gears 1.2.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * Reinforcement
  * PieMenu 1.7.5
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

Expressions

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-09:
does this work in 0.21.2?

krushia, 2024-07-09:
> does this work in 0.21.2?

Nope. Same results.

This also affects `_self.pi` (if a parameter named pi exists), `foo.pi`, and `<<foo>>.pi`. Interestingly it can't be entered at all in the expression editor, even with the conditional trick that _usually_ lets me enter naughty things in the "currently not evaluated" half.

[image]
[image]

luzpaz, 2025-01-25:
What's the way forward here ?

FEA-eng, 2025-10-02:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

FEA-eng, 2026-10-21:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 27.1.0dev.20261013 (Git shallow)
Build date: 2026/10/13 19:55:30
Build type: Release
Branch: (HEAD detached at 50daee777)
Hash: 50daee777321b5085accd1f57b57f1a7c1781cae
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```