FreeCAD/FreeCAD #15303: StackWalker is not in sync with upstream

krushia, 2024-07-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD bundles https://github.com/JochenKalmbach/StackWalker in the source code, but code contributions haven't been shared in either direction for at least 10 years. There was one failed attempt at #9675

Some upstream changes like https://github.com/JochenKalmbach/StackWalker/commit/8944b6d1f0366211ad02774f04855a5a0790e87e could benefit FreeCAD, and there have been fixes in FreeCAD that can be pushed upstream. Requires some careful auditing.

### Full version info

```shell
asdf
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

Not really a bug, but dunno where else to put these cleanup to-dos.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-07-27:
@wwmayer would you accept a PR for this ?

wwmayer, 2024-07-27:
https://github.com/FreeCAD/FreeCAD/pull/15639

closed 2024-07-30 by chennes