FreeCAD/FreeCAD #15305: Reliable crash in Data::ElementMap::addName exiting sketcher after adding constraint

johnsonm, 2024-07-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Rename [Frame.zip](https://github.com/user-attachments/files/16136170/Frame.zip) to Frame.FCStd and open it.

Open Sketch014 in SpindleFrame body:

[image]

Add a single point. It doesn't seem to matter where, but here I have added it to the left shorter line, accepting an automatic symmetric constraint:

[image]

Close the sketch, crash:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3c050) [0x7f80367d4050]
#1  0x7f8039afbaf6 in std::_Rb_tree<Data::MappedName, std::pair<Data::MappedName const, Data::IndexedName>, std::_Select1st<std::pair<Data::MappedName const, Data::IndexedName> >, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName> > >::_M_mbegin() const from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADApp.so+0xc
#2  0x7f8039afb9d8 in std::_Rb_tree<Data::MappedName, std::pair<Data::MappedName const, Data::IndexedName>, std::_Select1st<std::pair<Data::MappedName const, Data::IndexedName> >, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName> > >::_M_begin() from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADApp.so+0x18
#3  0x7f8039b211a2 in std::_Rb_tree<Data::MappedName, std::pair<Data::MappedName const, Data::IndexedName>, std::_Select1st<std::pair<Data::MappedName const, Data::IndexedName> >, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName> > >::lower_bound(Data::MappedName const&) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADApp.so+0x2c
#4  0x7f8039b1c5af in std::map<Data::MappedName, Data::IndexedName, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName> > >::lower_bound(Data::MappedName const&) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADApp.so+0x23
#5  0x7f8039b18650 in std::enable_if<std::is_constructible<std::pair<Data::MappedName const, Data::IndexedName>, std::pair<Data::MappedName, Data::IndexedName> >::value, std::pair<std::_Rb_tree_iterator<std::pair<Data::MappedName const, Data::IndexedName> >, bool> >::type std::map<Data::MappedName, Data::IndexedName, std::less<void>, std::allocator<std::pair<Data::MappedName const, Data::IndexedName> > >::insert<std::pair<Data::MappedName, Data::IndexedName> >(std::pair<Data::MappedName, Data::IndexedName>&&) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADApp.so+0x2c
#6  0x7f8039b0bb92 in Data::ElementMap::addName(Data::MappedName&, Data::IndexedName const&, QVector<App::StringIDRef> const&, bool, Data::IndexedName*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADApp.so+0x262
#7  0x7f8039b0c6de in Data::ElementMap::setElementName(Data::IndexedName const&, Data::MappedName const&, long, QVector<App::StringIDRef> const*, bool) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADApp.so+0x53e
#8  0x7f7ffb4ab6b7 in Data::ComplexGeoData::setElementName(Data::IndexedName const&, Data::MappedName const&, long, QVector<App::StringIDRef> const*, bool) from /var/home/johnsonm/d/FreeCAD/build/Mod/Sketcher/Sketcher.so+0x61
#9  0x7f7ffb46f1a1 in Sketcher::SketchObject::buildShape() from /var/home/johnsonm/d/FreeCAD/build/Mod/Sketcher/Sketcher.so+0x221
#10  0x7f7ffb46ecd7 in Sketcher::SketchObject::execute() from /var/home/johnsonm/d/FreeCAD/build/Mod/Sketcher/Sketcher.so+0x393
#11  0x7f8039829d3f in App::DocumentObject::recompute() from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADApp.so+0x75
#12  0x7f800a490059 in Part::Feature::recompute() from /var/home/johnsonm/d/FreeCAD/build/Mod/Part/Part.so+0x1d
#13  0x7f80397501ff in App::Document::_recomputeFeature(App::DocumentObject*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADApp.so+0x1eb
#14  0x7f803974db2e in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADApp.so+0x986
#15  0x7f80398e3c43 in App::DocumentPy::recompute(_object*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADApp.so+0x2c7
#16  0x7f80398db649 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADApp.so+0xf3
#17  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x1c9ba1) [0x7f8038024ba1]
#18  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x7d) [0x7f8037fd6e8d]
#19  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x4a63) [0x7f8037f64c63]
#20  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x3ccd3c) [0x7f8038227d3c]
#21  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyEval_EvalCode+0xc1) [0x7f80380c3891]
#22  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2b7359) [0x7f8038112359]
#23  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyRun_StringFlags+0x6b) [0x7f803811350b]
#24  0x7f803890bad1 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADBase.so+0xed
#25  0x7f803b4192d4 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0xc6
#26  0x7f803b418f8d in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0xfd
#27  0x7f803b41a11b in Gui::Command::updateActive() from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x3d
#28  0x7f7ff9238d74 in SketcherGui::ViewProviderSketch::unsetEdit(int) from /var/home/johnsonm/d/FreeCAD/build/Mod/Sketcher/SketcherGui.so+0x2e8
#29  0x7f803b9438af in Gui::ViewProvider::finishEditing() from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x31
#30  0x7f803b30b182 in Gui::Document::_resetEdit() from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x100
#31  0x7f803b253624 in Gui::Application::setEditDocument(Gui::Document*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x94
#32  0x7f803b30b07f in Gui::Document::resetEdit() from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x23
#33  0x7f803b381183 in Gui::DocumentPy::resetEdit(_object*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x4f
#34  0x7f803b37caad in Gui::DocumentPy::staticCallback_resetEdit(_object*, _object*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0xf3
#35  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x1c9ba1) [0x7f8038024ba1]
#36  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x7d) [0x7f8037fd6e8d]
#37  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x4a63) [0x7f8037f64c63]
#38  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x3ccd3c) [0x7f8038227d3c]
#39  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyEval_EvalCode+0xc1) [0x7f80380c3891]
#40  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2b7359) [0x7f8038112359]
#41  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(PyRun_StringFlags+0x6b) [0x7f803811350b]
#42  0x7f803890bad1 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADBase.so+0xed
#43  0x7f803b4192d4 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0xc6
#44  0x7f803b418f8d in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0xfd
#45  0x7f7ff92a1453 in SketcherGui::TaskDlgEditSketch::reject() from /var/home/johnsonm/d/FreeCAD/build/Mod/Sketcher/SketcherGui.so+0x19f
#46  0x7f803b8659c3 in Gui::TaskView::TaskView::reject() from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x9d
#47  0x7f803b8692fb in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::TaskView::TaskView::*)()>::call(void (Gui::TaskView::TaskView::*)(), Gui::TaskView::TaskView*, void**) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x6b
#48  0x7f803b868dc5 in void QtPrivate::FunctionPointer<void (Gui::TaskView::TaskView::*)()>::call<QtPrivate::List<>, void>(void (Gui::TaskView::TaskView::*)(), Gui::TaskView::TaskView*, void**) from /var/home/johnsonm/d/FreeCAD/build/lib/libFreeCADGui.so+0x42
[... cut ...]
  * fasteners 0.5.24
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
  * Assembly3 0.12.2
  * DynamicData 2.62.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-07-09:
Looks like another duplicate of #15102.

maxwxyz, 2024-07-10:
duplicate of #15102

closed 2024-07-10 by maxwxyz