FreeCAD/FreeCAD #15308: Reset preferences increases width of page

maxwxyz, 2024-07-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@kadet1090 FYI.

Resetting the Display preferences (entire Display group) increases the width of the preference, introducing a scrollbar at the bottom and hiding the settings :
Before reset:
[image]

After reset:
[image]

Also the first General tab has a scrollbar but no need to scroll with the default width:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git)
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-09:
Navigation tab also has an unnecessary scroll bar hiding the input boxes by default:
[image]

kadet1090, 2024-07-09:
Can you assign me? @maxwxyz

PR #15652 (2024-07-29): Gui: Cosmetic Fixes
This PR fixes various small cosmetic problems with preferences, like wrong padding, inconsistent alignment, horizontal scrollbar that should not be present etc.

Examples of before and after:
| Before | After |
| -- | -- |
|[image] | [image] |
| [image] | [image] |
| [image] |[image] |
| [image] | [image] |

Fixes: #14972 
Fixes: #14760 
Fixes: #15308