FreeCAD/FreeCAD #15310: [BIM] A door punches through more then 1 wall when the walls are close by

JustLinuxUser, 2024-07-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The simplest case:
Make a U shaped wall, with ~2m between the U branches, and put a standard door on one of the branches, The door is placed correctly, but it also punches a hole through the other branch of the U
[image]
In this case the wall was made from 1 sketch, distance between the branches of the U is 2m exactly, the wall settings are default, the door was placed on the inside wall

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git)
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Spain (es_ES)
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-07-09:
Indeed, you must lower the "Hole Depth" property. By default it's 1m https://wiki.freecad.org/Arch_Window

JustLinuxUser, 2024-07-09:
It was 0.0mm by default, but setting it to 0.5m did help, thanks!

closed 2024-07-09 by JustLinuxUser