FreeCAD/FreeCAD #15316: Freecad dosn't launch on deb 12 sid

dener3id, 2024-07-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I try to run the Appimage on Debian 12 and splash screen runs then hard crash.

### Full version info

```shell
'FreeCAD-0.21.2-Linux-x86_64.AppImage' Won't open to get to that dialog. IDK if core is right sub-project just assuming core is the core of the app and not a workbench.
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-10:
could you please try if the weekly AppImage works: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2024-07-11:
ping @dener3id

dener3id, 2024-07-11:
I'm testing now Seems to work ok so far

closed 2024-07-11 by maxwxyz