FreeCAD/FreeCAD #15320: Revolution up to face without function

TimTwix, 2024-07-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello,

if I create a datum and a simple sketch and choose revolution "up to face" nothing is changing. Freecad seems to take still the value from dimension.
[image]
[image]

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git) AppImage
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * OpenTheme 2024.5.16
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-10:
Please provide a demo file and steps how to reproduce. Are you trying to do the revolution up to the datum plane or a face?
@FlachyJoe FYI

Syres916, 2024-07-10:
This functionality is yet to coded see https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/PartDesign/App/FeatureRevolution.cpp#L183
Until that point the code shouldn't allow the user to select a Datum Plane IMHO.

TimTwix, 2024-07-13:
> Please provide a demo file and steps how to reproduce. Are you trying to do the revolution up to the datum plane or a face? @FlachyJoe FYI

I tried various references (e.g. created datums, origin planes) none of them working. 

Steps for reproduction:
-> create body
-> create a datum plane
-> create sketch
-> revolution -> up to face 

```
Python 3.11.9 | packaged by conda-forge | (main, Apr 19 2024, 18:53:30) [GCC 12.3.0] on linux
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> # Gui.runCommand('Start_Start',0)
>>> # Gui.runCommand('Std_Workbench',9)
>>> # Gui.runCommand('Std_Workbench',9)
>>> # Gui.runCommand('Std_ViewStatusBar',1)
>>> ### Begin command Std_New
>>> App.newDocument()
>>> # App.setActiveDocument("Unnamed")
>>> # App.ActiveDocument=App.getDocument("Unnamed")
>>> # Gui.ActiveDocument=Gui.getDocument("Unnamed")
>>> # Gui.activeDocument().activeView().viewDefaultOrientation()
>>> ### End command Std_New
>>> ### Begin command PartDesign_Body
>>> App.activeDocument().addObject('PartDesign::Body','Body')
>>> App.ActiveDocument.getObject('Body').Label = 'Body'
>>> # import PartDesignGui
>>> # Gui.activateView('Gui::View3DInventor', True)
>>> # Gui.activeView().setActiveObject('pdbody', App.activeDocument().Body)
>>> # Gui.Selection.clearSelection()
>>> # Gui.Selection.addSelection(App.ActiveDocument.Body)
>>> App.ActiveDocument.recompute()
>>> ### End command PartDesign_Body
>>> # Gui.Selection.addSelection('Unnamed','Body')
>>> # Gui.runCommand('Std_OrthographicCamera',1)
>>> # Gui.Selection.clearSelection()
>>> # Gui.Selection.addSelection('Unnamed','Body','Origin.XZ_Plane.')
>>> ### Begin command PartDesign_CompDatums
>>> App.getDocument('Unnamed').getObject('Body').newObject('PartDesign::Plane','DatumPlane')
>>> App.getDocument('Unnamed').getObject('DatumPlane').AttachmentSupport = [(App.getDocument('Unnamed').getObject('XZ_Plane'),'')]
>>> App.getDocument('Unnamed').getObject('DatumPlane').MapMode = 'FlatFace'
>>> App.activeDocument().recompute()
>>> # Gui.getDocument('Unnamed').setEdit(App.getDocument('Unnamed').getObject('Body'), 0, 'DatumPlane.')
>>> # import Show
>>> # from Show.Containers import isAContainer
>>> # _tv_DatumPlane = Show.TempoVis(App.ActiveDocument, tag= 'PartGui::TaskAttacher')
>>> # tvObj = App.getDocument('Unnamed').getObject('DatumPlane')
>>> # dep_features = _tv_DatumPlane.get_all_dependent(App.getDocument('Unnamed').getObject('Body'), 'DatumPlane.')
>>> # dep_features = [o for o in dep_features if not isAContainer(o)]
>>> # if tvObj.isDerivedFrom('PartDesign::CoordinateSystem'):
>>> # 	visible_features = [feat for feat in tvObj.InList if feat.isDerivedFrom('PartDesign::FeaturePrimitive')]
>>> # 	dep_features = [feat for feat in dep_features if feat not in visible_features]
>>> # 	del(visible_features)
>>> # _tv_DatumPlane.hide(dep_features)
>>> # del(dep_features)
>>> # if not tvObj.isDerivedFrom('PartDesign::CoordinateSystem'):
>>> # 		if len(tvObj.AttachmentSupport) > 0:
>>> # 			_tv_DatumPlane.show([lnk[0] for lnk in tvObj.AttachmentSupport])
>>> # del(tvObj)
>>> ### End command PartDesign_CompDatums
>>> # Gui.Selection.clearSelection()
>>> App.getDocument('Unnamed').getObject('DatumPlane').AttachmentOffset = App.Placement(App.Vector(0.0000000000, 0.0000000000, 0.0000000000),  App.Rotation(0.0000000000, 30.0000000000, 0.0000000000))
>>> App.getDocument('Unnamed').getObject('DatumPlane').MapReversed = False
>>> App.getDocument('Unnamed').getObject('DatumPlane').AttachmentSupport = [(App.getDocument('Unnamed').getObject('XZ_Plane'),'')]
>>> App.getDocument('Unnamed').getObject('DatumPlane').MapPathParameter = 0.000000
>>> App.getDocument('Unnamed').getObject('DatumPlane').MapMode = 'FlatFace'
>>> App.getDocument('Unnamed').getObject('DatumPlane').recompute()
>>> # Gui.getDocument('Unnamed').resetEdit()
>>> # _tv_DatumPlane.restore()
[... 89 lines cut ...]
>>> # App.getDocument('Unnamed').getObject('Revolution').ViewObject.ShapeAppearance=getattr(App.getDocument('Unnamed').getObject('Body').getLinkedObject(True).ViewObject,'ShapeAppearance',App.getDocument('Unnamed').getObject('Revolution').ViewObject.ShapeAppearance)
>>> # App.getDocument('Unnamed').getObject('Revolution').ViewObject.LineColor=getattr(App.getDocument('Unnamed').getObject('Body').getLinkedObject(True).ViewObject,'LineColor',App.getDocument('Unnamed').getObject('Revolution').ViewObject.LineColor)
>>> # App.getDocument('Unnamed').getObject('Revolution').ViewObject.PointColor=getattr(App.getDocument('Unnamed').getObject('Body').getLinkedObject(True).ViewObject,'PointColor',App.getDocument('Unnamed').getObject('Revolution').ViewObject.PointColor)
>>> # App.getDocument('Unnamed').getObject('Revolution').ViewObject.Transparency=getattr(App.getDocument('Unnamed').getObject('Body').getLinkedObject(True).ViewObject,'Transparency',App.getDocument('Unnamed').getObject('Revolution').ViewObject.Transparency)
>>> # App.getDocument('Unnamed').getObject('Revolution').ViewObject.DisplayMode=getattr(App.getDocument('Unnamed').getObject('Body').getLinkedObject(True).ViewObject,'DisplayMode',App.getDocument('Unnamed').getObject('Revolution').ViewObject.DisplayMode)
>>> # Gui.getDocument('Unnamed').setEdit(App.getDocument('Unnamed').getObject('Body'), 0, 'Revolution.')
>>> # Gui.Selection.clearSelection()
>>> ### End command PartDesign_Revolution
>>> # Gui.Selection.clearSelection()
>>> # Gui.Selection.addSelection('Unnamed','Body','DatumPlane.')
[... cut ...]
>>> App.getDocument('Unnamed').getObject('Revolution').Type = 3
>>> App.getDocument('Unnamed').getObject('Revolution').UpToFace = (App.getDocument("Unnamed").DatumPlane, [""])
>>> App.getDocument('Unnamed').recompute()
>>> ### Begin command Std_Save
>>> # Gui.SendMsgToActiveView("Save")
>>> App.getDocument("Unnamed").saveAs(u"/home/twix/Dokumente/FreeCAD/000_FREECAD/ISSUE_revolution_to_face.FCStd")
>>> ### End command Std_Save
>>> App.getDocument('Unnamed').recompute()
>>> # Gui.getDocument('Unnamed').resetEdit()
>>> 
```

maxwxyz, 2026-09-13:
@PaddleStroke FYI