FreeCAD/FreeCAD #15322: minor UI issues in 0.22Dev (windows 11, display resolution 1920x1080, orientation landscape)

kvacha13, 2024-07-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1) Windows taskbar overlapping Freecad page tabs. 
[image]
2) When you choose a certain dimensional constraint from the constraints dropdown box in sketcher, it does not keep the last selected constraint and every time you need to choose it again. 
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git)
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.65
  * Assembly3 0.12.2
  * Assembly4 0.50.13
  * CurvedShapes 1.0.9
  * Curves 0.6.39
  * fasteners 0.5.22
  * Manipulator 1.5.7
  * sheetmetal 0.4.20
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-10:
I cannot reproduce. Last tool stays active. Tested with all tools.
Maybe it's just a display issue that the mouse cursor changes back to default but the tool stays active, could you please test?
That's this issue: https://github.com/FreeCAD/FreeCAD/issues/12510 but there is already a fix.

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git)
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.40
  * dodo 1.0.1
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.22

```

kvacha13, 2024-07-10:
Hi Max,
Thank you for the reply.
Uploading a screen capture to show exactly what I meant. Unfortunately, the "dropdown box" has not been captured (no idea why). When you choose the constrain from the dropdown box, the toolbar icon does not change to the selected constrain icon (maybe it is supposed not to change in CAD 0.22)

https://github.com/FreeCAD/FreeCAD/assets/8162633/305c36b3-f053-4d3b-abe1-8601df2d8ba7

.

maxwxyz, 2024-07-10:
Is the issue that the toolbar icon does not change? That is by design, it should not change.

closed 2024-07-10 by maxwxyz