FreeCAD/FreeCAD #15325: Crash when invoking quit() in Python console if FreeCAD compiled with Qt 6

xtemp09, 2024-07-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If _FreeCAD_ compiled with _Qt 6_ has MDI windows opened, invocation of `quit()` in Python console causes segmentation fault with the following backtrace.

<details><summary>Backtrace</summary>

```cpp
Process 18351 stopped
* thread #1, name = 'FreeCAD', stop reason = signal SIGABRT
    frame #0: 0x00007ffff2e969fc libc.so.6`__GI___pthread_kill [inlined] __pthread_kill_implementation(no_tid=0, signo=6, threadid=140737238932480) at pthread_kill.c:44:76
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGABRT
  * frame #0: 0x00007ffff2e969fc libc.so.6`__GI___pthread_kill [inlined] __pthread_kill_implementation(no_tid=0, signo=6, threadid=140737238932480) at pthread_kill.c:44:76
    frame #1: 0x00007ffff2e969b0 libc.so.6`__GI___pthread_kill [inlined] __pthread_kill_internal(signo=6, threadid=140737238932480) at pthread_kill.c:78:10
    frame #2: 0x00007ffff2e969b0 libc.so.6`__GI___pthread_kill(threadid=140737238932480, signo=6) at pthread_kill.c:89:10
    frame #3: 0x00007ffff2e42476 libc.so.6`__GI_raise(sig=6) at raise.c:26:13
    frame #4: 0x00007ffff2e287f3 libc.so.6`__GI_abort at abort.c:79:7
    frame #5: 0x00007ffff7894a86 libFreeCADGui.so`messageHandler(type=QtFatalMsg, context=<unavailable>, msg=0x00007fffffffc650) at Application.cpp:1823:14
    frame #6: 0x00007ffff3702aa7 libQt6Core.so.6`qt_message(QtMsgType, QMessageLogContext const&, char const*, __va_list_tag*) + 183
    frame #7: 0x00007ffff36ba1df libQt6Core.so.6`QMessageLogger::fatal(char const*, ...) const + 165
    frame #8: 0x00007ffff36b935c libQt6Core.so.6`qt_assert_x(char const*, char const*, char const*, int) + 96
    frame #9: 0x00007ffff7a9515a libFreeCADGui.so`void QtPrivate::assertObjectType<Gui::MainWindow>(o=0x00007fffffffd680) at qobjectdefs_impl.h:129:9
    frame #10: 0x00007ffff7d8e1b0 libFreeCADGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<0>, QtPrivate::List<QMdiSubWindow*>, void, void (Gui::MainWindow::*)(QMdiSubWindow*)>::call(f=(libFreeCADGui.so`Gui::MainWindow::onWindowActivated(QMdiSubWindow*) at MainWindow.cpp:1317:1), o=0x00007fffffffd680, arg=0x00007fffffffc930) at qobjectdefs_impl.h:144:34
    frame #11: 0x00007ffff7d8e212 libFreeCADGui.so`QtPrivate::QCallableObject<void (Gui::MainWindow::*)(QMdiSubWindow*), QtPrivate::List<QMdiSubWindow*>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) [inlined] void QtPrivate::FunctionPointer<void (Gui::MainWindow::*)(QMdiSubWindow*)>::call<QtPrivate::List<QMdiSubWindow*>, void>(arg=<unavailable>, o=<unavailable>, f=<unavailable>) at qobjectdefs_impl.h:182:95
    frame #12: 0x00007ffff7d8e20d libFreeCADGui.so`QtPrivate::QCallableObject<void (Gui::MainWindow::*)(QMdiSubWindow*), QtPrivate::List<QMdiSubWindow*>, void>::impl(which=<unavailable>, this_=<unavailable>, r=<unavailable>, a=<unavailable>, ret=<unavailable>) at qobjectdefs_impl.h:553:53
    frame #13: 0x00007ffff37def50 libQt6Core.so.6`void doActivate<false>(QObject*, int, void**) + 1744
    frame #14: 0x00007ffff4d3f7c3 libQt6Widgets.so.6`QMdiArea::subWindowActivated(QMdiSubWindow*) + 67
    frame #15: 0x00007ffff4d45dcb libQt6Widgets.so.6`QMdiAreaPrivate::_q_processWindowStateChanged(QFlags<Qt::WindowState>, QFlags<Qt::WindowState>) + 299
    frame #16: 0x00007ffff37df1df libQt6Core.so.6`void doActivate<false>(QObject*, int, void**) + 2399
    frame #17: 0x00007ffff4d4be59 libQt6Widgets.so.6`QMdiSubWindow::windowStateChanged(QFlags<Qt::WindowState>, QFlags<Qt::WindowState>) + 89
    frame #18: 0x00007ffff4d55e00 libQt6Widgets.so.6`QMdiSubWindowPrivate::setActive(bool, bool) + 752
    frame #19: 0x00007ffff4d480d2 libQt6Widgets.so.6`QMdiArea::event(QEvent*) + 114
    frame #20: 0x00007ffff4b8b345 libQt6Widgets.so.6`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 133
    frame #21: 0x00007ffff7948254 libFreeCADGui.so`Gui::GUIApplication::notify(this=0x00007fffffffd6b0, receiver=0x0000555555b317b0, event=0x00007fffffffcee0) at GuiApplication.cpp:83:40
    frame #22: 0x00007ffff377b398 libQt6Core.so.6`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 344
    frame #23: 0x00007ffff4be1150 libQt6Widgets.so.6`QWidgetPrivate::hideChildren(bool) + 416
    frame #24: 0x00007ffff4be1757 libQt6Widgets.so.6`QWidgetPrivate::hide_helper() + 327
    frame #25: 0x00007ffff4be5fda libQt6Widgets.so.6`QWidgetPrivate::setVisible(bool) + 922
    frame #26: 0x00007ffff4be1983 libQt6Widgets.so.6`QWidgetPrivate::handleClose(QWidgetPrivate::CloseMode) + 259
    frame #27: 0x00007ffff4bf789f libQt6Widgets.so.6`QWidgetWindow::closeEvent(QCloseEvent*) + 63
    frame #28: 0x00007ffff402d152 libQt6Gui.so.6`QWindow::event(QEvent*) + 1618
    frame #29: 0x00007ffff4b8b345 libQt6Widgets.so.6`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 133
    frame #30: 0x00007ffff7948254 libFreeCADGui.so`Gui::GUIApplication::notify(this=0x00007fffffffd6b0, receiver=0x000055555ba822d0, event=0x00007fffffffd500) at GuiApplication.cpp:83:40
    frame #31: 0x00007ffff377b398 libQt6Core.so.6`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 344
    frame #32: 0x00007ffff3fcce81 libQt6Gui.so.6`QGuiApplicationPrivate::processCloseEvent(QWindowSystemInterfacePrivate::CloseEvent*) + 177
    frame #33: 0x00007ffff403b38c libQt6Gui.so.6`bool QWindowSystemInterface::handleCloseEvent<QWindowSystemInterface::SynchronousDelivery>(QWindow*) + 204
    frame #34: 0x00007ffff4026bf0 libQt6Gui.so.6`QWindow::close() + 96
    frame #35: 0x00007ffff4be204e libQt6Widgets.so.6`QWidget::~QWidget() + 1150
    frame #36: 0x00007ffff7d84214 libFreeCADGui.so`Gui::MainWindow::~MainWindow(this=0x00007fffffffd680) at MainWindow.cpp:530:1
    frame #37: 0x00007ffff78983f5 libFreeCADGui.so`Gui::Application::runApplication() at Application.cpp:2163:1
    frame #38: 0x0000555555570ab7 FreeCAD`main(argc=<unavailable>, argv=<unavailable>) at MainGui.cpp:300:45
    frame #39: 0x00007ffff2e29d90 libc.so.6`__libc_start_call_main(main=(FreeCAD`main at MainGui.cpp:107:1), argc=1, argv=0x00007fffffffdd48) at libc_start_call_main.h:58:16
    frame #40: 0x00007ffff2e29e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:107:1), argc=1, argv=0x00007fffffffdd48, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdd38) at libc-start.c:392:3
    frame #41: 0x00005555555702e5 FreeCAD`_start + 37
(lldb)
```

</details>

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38045 (Git)
Build type: Debug
Branch: RFC_shiboken6fix2
Hash: de3eb7578f73c90447ff1ba878077a805f9a8a0e
Python 3.10.12, Qt 6.7.1, Coin 4.0.0, Vtk , OCC 7.5.1
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pieterhijma, 2024-07-10:
Some more information: it also happens in the main branch:

```
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38098 (Git)
Build type: DEBUG
Branch: main
Hash: 5b9b611f8ec223a8bc3c8739fee92bea3b4a5705
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk , OCC 7.7.2
Locale: English/United States (en_US)
```

It doesn't happen when there is no MDI view open. Opening a file or have the start workbench open triggers this bug.

To reproduce:
- make sure there is an MDI view open (file or Start workbench)
- type `quit()` in the console
- answer yes

The following failed assertion is shown:

```
ASSERT failure in Gui::MainWindow: "Called object is not of the correct type (class destructor may have already run)", file /usr/include/qt6/QtCore/qobjectdefs_impl.h, line 130
```

<details>
<summary>My stacktrace:</summary>

```
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007ffff08a8eb3 in __pthread_kill_internal (threadid=<optimized out>, signo=6) at pthread_kill.c:78
#2  0x00007ffff0850a30 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff08384c3 in __GI_abort () at abort.c:79
#4  0x00007ffff66593db in messageHandler (type=QtFatalMsg, context=..., msg=...) at /home/pieter/current/ondsel/freecad-main/src/Gui/Application.cpp:1823
#5  0x00007ffff0edcf24 in qt_message_print (msgType=QtFatalMsg, context=..., message=...) at /usr/src/debug/qt6-base/qtbase/src/corelib/global/qlogging.cpp:2018
#6  0x00007ffff0edd07e in qt_message(QtMsgType, const QMessageLogContext &, const char *, typedef __va_list_tag __va_list_tag *) (msgType=msgType@entry=QtFatalMsg, context=..., msg=msg@entry=0x7ffff1221fb8 "ASSERT failure in %s: \"%s\", file %s, line %d", ap=ap@entry=0x7fffffffae00) at /usr/src/debug/qt6-base/qtbase/src/corelib/global/qlogging.cpp:371
#7  0x00007ffff0e8cde7 in QMessageLogger::fatal (this=this@entry=0x7fffffffaee8, msg=msg@entry=0x7ffff1221fb8 "ASSERT failure in %s: \"%s\", file %s, line %d") at /usr/src/debug/qt6-base/qtbase/src/corelib/global/qlogging.cpp:889
#8  0x00007ffff0e8ce98 in qt_assert_x (where=<optimized out>, what=<optimized out>, file=<optimized out>, line=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/global/qassert.cpp:77
#9  0x00007ffff6afbdec in QtPrivate::assertObjectType<Gui::MainWindow> (o=0x7fffffffc180) at /usr/include/qt6/QtCore/qobjectdefs_impl.h:129
#10 0x00007ffff70d28ae in QtPrivate::FunctorCall<QtPrivate::IndexesList<0>, QtPrivate::List<QMdiSubWindow*>, void, void (Gui::MainWindow::*)(QMdiSubWindow*)>::call (f=(void (Gui::MainWindow::*)(class Gui::MainWindow * const, class QMdiSubWindow *)) 0x7ffff70b4af0 <Gui::MainWindow::onWindowActivated(QMdiSubWindow*)>, o=0x7fffffffc180, arg=0x7fffffffb160) at /usr/include/qt6/QtCore/qobjectdefs_impl.h:144
#11 0x00007ffff70d284d in QtPrivate::FunctionPointer<void (Gui::MainWindow::*)(QMdiSubWindow*)>::call<QtPrivate::List<QMdiSubWindow*>, void> (f=(void (Gui::MainWindow::*)(class Gui::MainWindow * const, class QMdiSubWindow *)) 0x7ffff70b4af0 <Gui::MainWindow::onWindowActivated(QMdiSubWindow*)>, o=0x7fffffffc180, arg=0x7fffffffb160) at /usr/include/qt6/QtCore/qobjectdefs_impl.h:182
#12 0x00007ffff70d2775 in QtPrivate::QCallableObject<void (Gui::MainWindow::*)(QMdiSubWindow*), QtPrivate::List<QMdiSubWindow*>, void>::impl (which=1, this_=0x555555d85ee0, r=0x7fffffffc180, a=0x7fffffffb160, ret=0x0) at /usr/include/qt6/QtCore/qobjectdefs_impl.h:553
#13 0x00007ffff0fa16f7 in QtPrivate::QSlotObjectBase::call (this=<optimized out>, r=<optimized out>, a=<optimized out>, this=<optimized out>, r=<optimized out>, a=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qobjectdefs_impl.h:469
#14 doActivate<false> (sender=<optimized out>, signal_index=<optimized out>, argv=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qobject.cpp:4086
#15 0x00007ffff20b62d9 in QMdiArea::subWindowActivated (this=0x555555c408b0, _t1=0x0) at /usr/src/debug/qt6-base/build/src/widgets/Widgets_autogen/include/moc_qmdiarea.cpp:346
#16 QMdiAreaPrivate::resetActiveWindow (this=0x555555c1b740, deactivatedWindow=0x55555858d1e0) at /usr/src/debug/qt6-base/qtbase/src/widgets/widgets/qmdiarea.cpp:1060
#17 QMdiAreaPrivate::_q_processWindowStateChanged (this=0x555555c1b740, oldState=..., newState=...) at /usr/src/debug/qt6-base/qtbase/src/widgets/widgets/qmdiarea.cpp:718
#18 QMdiArea::qt_static_metacall (_o=<optimized out>, _c=<optimized out>, _id=<optimized out>, _a=<optimized out>) at /usr/src/debug/qt6-base/build/src/widgets/Widgets_autogen/include/moc_qmdiarea.cpp:246
#19 0x00007ffff0fa194a in doActivate<false> (sender=<optimized out>, signal_index=<optimized out>, argv=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qobject.cpp:4098
#20 0x00007ffff20c025c in QMdiSubWindow::windowStateChanged (this=<optimized out>, _t1=..., _t2=...) at /usr/src/debug/qt6-base/build/src/widgets/Widgets_autogen/include/moc_qmdisubwindow.cpp:241
#21 QMdiSubWindowPrivate::setActive (this=0x55555961c5b0, activate=<optimized out>, changeFocus=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/widgets/widgets/qmdisubwindow.cpp:1450
#22 0x00007ffff20b4cc5 in QMdiAreaPrivate::setActive (active=false, changeFocus=false, this=<optimized out>, subWindow=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/widgets/widgets/qmdiarea_p.h:219
[... cut ...]
#49 0x00007ffff1f33816 in QWidget::~QWidget (this=<optimized out>, this=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qwidget.cpp:1510
#50 0x00007ffff70b5203 in Gui::MainWindow::~MainWindow (this=0x7fffffffc180) at /home/pieter/current/ondsel/freecad-main/src/Gui/MainWindow.cpp:530
#51 0x00007ffff6659e24 in Gui::Application::runApplication () at /home/pieter/current/ondsel/freecad-main/src/Gui/Application.cpp:2163
#52 0x000055555557737d in main (argc=1, argv=0x7fffffffd5f8) at /home/pieter/current/ondsel/freecad-main/src/Main/MainGui.cpp:300
```
</detail>

xtemp09, 2024-07-12:
_Qt 6.3_ introduces segmentation fault via assert if a _QObject_ receives a signal during destructor run. [^1]

In _QTBUG-105473_ Thiago Macieira explains that this is not a bug, but a feature, [^2] meaning that, _FreeCAD_ compiled with a debug version of _Qt_ is going to sporadically crash.

Should the issue be closed?

[^1]: https://github.com/qt/qtbase/commit/0e72a846d379ba02ff80ecac2526640a05b872b6
[^2]: https://bugreports.qt.io/browse/QTBUG-105473

----

Here is a better backtrace.

<details><summary>Backtrace</summary>

```cpp
System exit
ASSERT failure in Gui::MainWindow: "Called object is not of the correct type (class destructor may have already run)", file /usr/local/Qt-6.7.2/include/QtCore/qobjectdefs_impl.h, line 129
Process 1684 stopped
* thread #1, name = 'FreeCAD', stop reason = signal SIGABRT
    frame #0: 0x00007ffff2e969fc libc.so.6`__GI___pthread_kill [inlined] __pthread_kill_implementation(no_tid=0, signo=6, threadid=140737266350912) at pthread_kill.c:44:76
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGABRT
  * frame #0: 0x00007ffff2e969fc libc.so.6`__GI___pthread_kill [inlined] __pthread_kill_implementation(no_tid=0, signo=6, threadid=140737266350912) at pthread_kill.c:44:76
    frame #1: 0x00007ffff2e969b0 libc.so.6`__GI___pthread_kill [inlined] __pthread_kill_internal(signo=6, threadid=140737266350912) at pthread_kill.c:78:10
    frame #2: 0x00007ffff2e969b0 libc.so.6`__GI___pthread_kill(threadid=140737266350912, signo=6) at pthread_kill.c:89:10
    frame #3: 0x00007ffff2e42476 libc.so.6`__GI_raise(sig=6) at raise.c:26:13
    frame #4: 0x00007ffff2e287f3 libc.so.6`__GI_abort at abort.c:79:7
    frame #5: 0x00007ffff7893e82 libFreeCADGui.so`messageHandler(type=QtFatalMsg, context=<unavailable>, msg=0x00007fffffffc090) at Application.cpp:1823:14
    frame #6: 0x00007ffff36d78cd libQt6Core.so.6`qt_message_print(msgType=QtFatalMsg, context=0x00007fffffffc1c8, message=0x00007fffffffc090) at qlogging.cpp:2018:59
    frame #7: 0x00007ffff36d7b74 libQt6Core.so.6`qt_message(msgType=QtFatalMsg, context=0x00007fffffffc1c8, msg="", ap=0x00007fffffffc0e0) at qlogging.cpp:371:21
    frame #8: 0x00007ffff36d8ba5 libQt6Core.so.6`QMessageLogger::fatal(this=0x00007fffffffc1c8, msg="") const at qlogging.cpp:889:5
    frame #9: 0x00007ffff36d166e libQt6Core.so.6`qt_assert_x(where=<unavailable>, what="", file="", line=129) at qassert.cpp:77:19
    frame #10: 0x00007ffff7a95ad2 libFreeCADGui.so`void QtPrivate::assertObjectType<Gui::MainWindow>(o=0x00007fffffffd680) at qobjectdefs_impl.h:129:9
    frame #11: 0x00007ffff7d9070e libFreeCADGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<0>, QtPrivate::List<QMdiSubWindow*>, void, void (Gui::MainWindow::*)(QMdiSubWindow*)>::call(f=(libFreeCADGui.so`Gui::MainWindow::onWindowActivated(QMdiSubWindow*) at MainWindow.cpp:1320:1), o=0x00007fffffffd680, arg=0x00007fffffffc380) at qobjectdefs_impl.h:144:34
    frame #12: 0x00007ffff7d90770 libFreeCADGui.so`QtPrivate::QCallableObject<void (Gui::MainWindow::*)(QMdiSubWindow*), QtPrivate::List<QMdiSubWindow*>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) [inlined] void QtPrivate::FunctionPointer<void (Gui::MainWindow::*)(QMdiSubWindow*)>::call<QtPrivate::List<QMdiSubWindow*>, void>(arg=<unavailable>, o=<unavailable>, f=<unavailable>) at qobjectdefs_impl.h:182:95
    frame #13: 0x00007ffff7d9076b libFreeCADGui.so`QtPrivate::QCallableObject<void (Gui::MainWindow::*)(QMdiSubWindow*), QtPrivate::List<QMdiSubWindow*>, void>::impl(which=<unavailable>, this_=<unavailable>, r=<unavailable>, a=<unavailable>, ret=<unavailable>) at qobjectdefs_impl.h:553:53
    frame #14: 0x00007ffff3798962 libQt6Core.so.6`void doActivate<false>(QObject*, int, void**) [inlined] QtPrivate::QSlotObjectBase::call(a=0x00007fffffffc380, r=0x00007fffffffd680, this=0x0000555555b399c0) at qobjectdefs_impl.h:469:57
    frame #15: 0x00007ffff3798946 libQt6Core.so.6`void doActivate<false>(sender=0x0000555555b4b290, signal_index=<unavailable>, argv=<unavailable>) at qobject.cpp:4086:30
    frame #16: 0x00007ffff378dedc libQt6Core.so.6`QMetaObject::activate(sender=0x0000555555b4b290, m=0x0000000000759680, local_signal_index=0, argv=0x00007fffffffc380) at qobject.cpp:4146:26
    frame #17: 0x00007ffff4d27398 libQt6Widgets.so.6`QMdiArea::subWindowActivated(this=0x0000555555b4b290, _t1=<unavailable>) at moc_qmdiarea.cpp:349:26
    frame #18: 0x00007ffff4d2964e libQt6Widgets.so.6`QMdiAreaPrivate::resetActiveWindow(this=0x0000555555b026c0, deactivatedWindow=0x0000555555e644c0) at qmdiarea.cpp:1060:35
    frame #19: 0x00007ffff4d2c8bf libQt6Widgets.so.6`QMdiAreaPrivate::_q_processWindowStateChanged(this=0x0000555555b026c0, oldState=(i = 10), newState=(i = 2)) at qmdiarea.cpp:718:26
    frame #20: 0x00007ffff4d2ca47 libQt6Widgets.so.6`QMdiArea::qt_static_metacall(_o=0x0000555555b4b290, _c=<unavailable>, _id=10, _a=0x00007fffffffc580) at moc_qmdiarea.cpp:247:60
    frame #21: 0x00007ffff3798809 libQt6Core.so.6`void doActivate<false>(sender=0x0000555555e644c0, signal_index=<unavailable>, argv=<unavailable>) at qobject.cpp:4098:33
    frame #22: 0x00007ffff378dedc libQt6Core.so.6`QMetaObject::activate(sender=0x0000555555e644c0, m=0x0000000000759e20, local_signal_index=0, argv=0x00007fffffffc580) at qobject.cpp:4146:26
    frame #23: 0x00007ffff4d350ee libQt6Widgets.so.6`QMdiSubWindow::windowStateChanged(this=<unavailable>, _t1=(i = 10), _t2=(i = 2)) at moc_qmdisubwindow.cpp:244:26
[... cut ...]
    frame #59: 0x00007ffff2e29d90 libc.so.6`__libc_start_call_main(main=(FreeCAD`main at MainGui.cpp:107:1), argc=1, argv=0x00007fffffffdd48) at libc_start_call_main.h:58:16
    frame #60: 0x00007ffff2e29e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:107:1), argc=1, argv=0x00007fffffffdd48, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdd38) at libc-start.c:392:3
    frame #61: 0x00005555555702a5 FreeCAD`_start + 37
(lldb)
``` 

</details>

xtemp09, 2024-07-12:
The commit 292f298c4206fa29690a839c2f3c42f2a0d54e7f fixes the issue.

luzpaz, 2024-07-22:
@xtemp09 can this ticket then be closed ?

xtemp09, 2024-07-23:
I'm not Vanga, but in the future many issues will have similar backtrace.

closed 2024-07-23 by xtemp09

pieterhijma, 2024-07-23:
I disagree that this issue can be closed. The referenced post above only states (and I fully agree) that occurrence of this segfault is not a bug in Qt code. At some point the developers decided that Qt will report and crash if you call a slot on the main window after deleting the main window.

So, there is a bug in FreeCAD code (the main window still receives signals after it has been deleted) and I believe we should fix it.

I propose to reopen this issue and I hope @xtemp09 can submit the commit as a PR with the goal to close this issue. To be sure, let's ping @wwmayer to ask if https://github.com/FreeCAD/FreeCAD/commit/292f298c4206fa29690a839c2f3c42f2a0d54e7f is indeed the right way to fix this issue.

wwmayer, 2024-07-23:
I cannot confirm the crash, neither with Qt5 nor with Qt6. But if this change fixes the crash for you it's fine to make a PR from it.