FreeCAD/FreeCAD #15329: build error on macos 12.16 arm64 architecture offending file, TopoShape.h

ipatch, 2024-07-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

i think the TopoShape.h is missing a "include" and by include i mean,

```
#include <unordered_map>
```

after adding that line to the file `TopoShape.h` the build continues without a _fatal error_

i'm using macos 12.6 on a computer provided by compile farm.

and am aware that macos can be probelmatic, but just thought i'd share this finding. i haven't attempted to compile on a linux box ie. debian or arch.

### Full version info

```shell
NA
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ipatch, 2024-07-09:
ohh and i'm attempting to build freecad from the below revision,

https://github.com/FreeCAD/FreeCAD/tree/dfe0c5d4ee4841f11e62da1104e322e95834aba2

luzpaz, 2024-07-11:
Added to Milestone since it's a potential packaging issue and should be fixed pre release.

ipatch, 2024-07-11:
hey for what it's worth, i run arch linux, btw™️ 🤡 on my local 2015 mbp. and when i ran

```
brew install freecad@0.21.2_py310 -v --HEAD
```

i ran into this same issue, so not entirely sure it's just mac packing issue or a greater homebrew issue. i've never build freecad with conda so not sure if this issue would arise in a conda, but this issue not freecad specific, but seems more of a general c++ issue from what i can tell. (maybe a c++ issue that is related to homebrew specifically however due to maybe the compilers and libraries being used 🤔.

but for reference here is similar general c++ error i came across when searching for a fix.

https://stackoverflow.com/questions/15733163/c-error-unordered-map-does-not-name-a-type

and the fix for my local development install of freecad was to include the missing header in the offending file

https://github.com/FreeCAD/homebrew-freecad/blob/master/patches/freecad%400.21.2_py310-HEAD-toposhape.h-fix.patch

which will hopefully all the development builds work with `brew install freecad --HEAD` once I resolve the below PR with the homebrew-freecad tap

https://github.com/FreeCAD/homebrew-freecad/pull/535

luzpaz, 2024-09-24:
@ipatch would you like to make a PR against main ?

closed 2024-09-25 by luzpaz