FreeCAD/FreeCAD #15330: Empty element map if Refine is enabled

wwmayer, 2024-07-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://github.com/FreeCAD/FreeCAD/pull/15096#issuecomment-2200872127

```
doc = App.newDocument()
box1 = doc.addObject("Part::Box","Box")
box2 = doc.addObject("Part::Box","Box")
fuse = doc.addObject("Part::MultiFuse","Fusion")
doc.Fusion.Refine = True  # activate refinement
doc.Fusion.Shapes = [box1, box2]
doc.recompute()
print (fuse.Shape.ElementMapSize)
```
ElementMapSize is 0

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38090 (Git)
Build type: debug
Branch: main
Hash: fa2741fd292909eebcad965ebcdf3599bd20f25b
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-07-30:
Fix needs to happen pre- or post-v1.0 ?

wwmayer, 2024-07-30:
IMO it should be "pre".

PR #15897 (2024-08-18): Toponaming:  Fuse call makeElementRefine; remove deprecated refineSha…
…peIfActive  fix #15330
closed 2024-08-18 by chennes