FreeCAD/FreeCAD #15331: UI nit: Tab bar is truncating tab titles too early due to new padding

depthoffocus, 2024-07-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I like the tab bar a _lot_, it really makes the app feel more uniform and less workbench-hide-and-seek, somehow. 

But I think the recently tweaked presentation in 38079 is favouring the padding over the text somewhat unreasonably. 

I have my menu set up (on a 13" MacBook Pro) so that some of the universal panels are to the right of it on the top row in each workbench, but that means shrinking it down a little.

Here's an example where the ellipsis and padding is taking away the space where the workbench titles would clearly fit.

I think the right padding might be a bit too assertive, but is it possible if not to _fade out_ the text rather than replace it with an ellipsis? This could retain readability for a lot longer. Though it may be more difficult to support in themes, I guess.
 
[image]

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git)
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.90.0
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * Curves 0.6.40
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-10:
@kadet1090 @MisterMakerNL FYI

depthoffocus, 2024-07-10:
Incidentally if I switch to another theme, say the new Light theme, and then switch back, the tab bar then looks like this instead:

[image]

kadet1090, 2024-07-10:
Hm, I suspect that this issue is not present on previous weekly, i.e. 38078 ?

depthoffocus, 2024-07-10:
> Hm, I suspect that this issue is not present on previous weekly, i.e. 38078 ?

It changed with the new version, yes.

luzpaz, 2024-07-17:
bumping for visibility

MisterMakerNL, 2024-07-20:
Maybe is https://github.com/FreeCAD/FreeCAD/issues/15533 is related?

depthoffocus, 2024-07-30:
FWIW I think this behaviour is still somewhat inconsistent in 38334 and 38375, though the tab bar's presentation is now more like in my second screenshot above.

FreeCAD Light:
[image]

FreeCAD Dark:
[image]

FreeCAD Classic:
[image]

luzpaz, 2024-09-14:
bumping again

maxwxyz, 2025-07-11:
@MisterMakerNL is this addressed in your PR?

closed 2025-07-15 by maxwxyz