FreeCAD/FreeCAD #15336: Crash on recompute in recent weekly build

CapnKernel, 2024-07-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a .FCStd file which reliably causes a segfault in a weekly build.

Weekly build 37819 does not crash, weekly build 38079 does.

# Instructions for reproducing:

- Load supplied .FCStd file: `~/AppImage/FreeCAD_weekly-builds-38079-conda-Linux-x86_64-py311.AppImage edges2b.FCStd`
- Press `Ctrl-R` to refresh document.
- App crashes with the following backtrace:
```
Sketcher External Geometry without ExternalGeometryExtension: cfe4e96f-1d7a-4856-8e69-73d3cc6762e0 
Sketcher External Geometry without ExternalGeometryExtension: 3c7ba97e-60b2-47cd-bc78-14c932081dfb 
Sketcher External Geometry without ExternalGeometryExtension: a3db600e-9682-46ed-a318-4f5e6d511ed9 
Sketcher External Geometry without ExternalGeometryExtension: 0ecc134d-eb82-4559-a50f-085da281592a 
Sketcher External Geometry without ExternalGeometryExtension: 181359d6-7fd9-4eb3-8261-0c5bd15d8780 
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x3dbb0) [0x7f07ab85fbb0]
#1  0x7f07ada7eead in Data::ElementMap::addName(Data::MappedName&, Data::IndexedName const&, QVector<App::StringIDRef> const&, bool, Data::IndexedName*) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libFreeCADApp.so+0xed
#2  0x7f07ada7fc97 in Data::ElementMap::setElementName(Data::IndexedName const&, Data::MappedName const&, long, QVector<App::StringIDRef> const*, bool) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libFreeCADApp.so+0x297
#3  0x7f07594f91b6 in Sketcher::SketchObject::buildShape() from /tmp/.mount_FreeCAETIJLg/usr/lib/Sketcher.so+0x236
#4  0x7f07594fa1ea in Sketcher::SketchObject::execute() from /tmp/.mount_FreeCAETIJLg/usr/lib/Sketcher.so+0xda
#5  0x7f07ad85d552 in App::DocumentObject::recompute() from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libFreeCADApp.so+0xc2
#6  0x7f07668a23cc in Part::Feature::recompute() from /tmp/.mount_FreeCAETIJLg/usr/lib/./Part.so+0xc
#7  0x7f07ad83773c in App::Document::_recomputeFeature(App::DocumentObject*) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libFreeCADApp.so+0x32c
#8  0x7f07ad843ae5 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libFreeCADApp.so+0x625
#9  0x7f07ad8c21ae in App::DocumentPy::recompute(_object*) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libFreeCADApp.so+0x2ee
#10  0x7f07ad8c22c4 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libFreeCADApp.so+0x24
#11  /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x7f07ae006068]
#12  /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7f07adfe5b5c]
#13  /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7f07adff1532]
#14  /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7f07ae0abd1d]
#15  /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7f07ae0ab794]
#16  /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7f07ae0cabf3]
#17  /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7f07ae0c669a]
#18  /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7f07ae0ba9c9]
#19  0x7f07ad33d48c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libFreeCADBase.so+0x6c
#20  0x7f07ae987924 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libFreeCADGui.so+0x94
#21  0x7f07ae987ab3 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libFreeCADGui.so+0xe3
#22  0x7f07ae995ac1 in StdCmdRefresh::activated(int) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libFreeCADGui.so+0x71
#23  0x7f07ae99167e in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libFreeCADGui.so+0xde
#24  0x7f07ae991c04 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libFreeCADGui.so+0x134
#25  /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7f07abceb6f2]
#26  0x7f07acb6a462 in QAction::triggered(bool) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Widgets.so.5+0x32
#27  0x7f07acb6ce2b in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Widgets.so.5+0xab
#28  /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Widgets.so.5(+0x25d29a) [0x7f07acc5d29a]
#29  0x7f07acc5d3d7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Widgets.so.5+0xe7
#30  0x7f07acd4610a in QToolButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Widgets.so.5+0xa
#31  0x7f07acbaec08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Widgets.so.5+0x178
#32  0x7f07acb703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#33  0x7f07acb78262 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#34  0x7f07ae93cc29 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libFreeCADGui.so+0x99
#35  0x7f07abcb6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Core.so.5+0x108
#36  0x7f07acb763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#37  /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x7f07acbc6e6f]
#38  /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7f07acbca0c9]
#39  0x7f07acb703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#40  0x7f07ae93cc29 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libFreeCADGui.so+0x99
#41  0x7f07abcb6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Core.so.5+0x108
#42  0x7f07ac3353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Gui.so.5+0x84c
#43  0x7f07ac30b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Gui.so.5+0xac
#44  /tmp/.mount_FreeCAETIJLg/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7f07994adbea]
#45  /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7f07aa510ad7]
#46  /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7f07aa513ac7]
#47  /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f07aa5140ce]
#48  0x7f07abd0aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Core.so.5+0x66
#49  0x7f07abcb570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Core.so.5+0x11b
#50  0x7f07abcbd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libQt5Core.so.5+0x82
#51  0x7f07ae89f362 in Gui::Application::runApplication() from /tmp/.mount_FreeCAETIJLg/usr/bin/../lib/libFreeCADGui.so+0x972
#52  /tmp/.mount_FreeCAETIJLg/usr/bin/freecad(+0x7620) [0x560acb8cc620]
#53  /lib64/libc.so.6(+0x27b8a) [0x7f07ab849b8a]
#54  /lib64/libc.so.6(__libc_start_main+0x8b) [0x7f07ab849c4b]
#55  /tmp/.mount_FreeCAETIJLg/usr/bin/freecad(+0x7981) [0x560acb8cc981]
```

Loading `edges2b.FCStd` and refreshing in weekly build 37819 does not crash.

### Full version info

```shell
OS: Fedora Linux 38 (Thirty Eight) (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git) AppImage
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Installed mods: 
  * Curves 0.6.40
  * kicadStepUpMod 11.2.4
  * sheetmetal 0.4.22
  * ThreadProfile 1.90.0
```

### Subproject(s) affected?

None

### Anything else?

[edges2-crasher.zip](https://github.com/user-attachments/files/16156406/edges2-crasher.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-07-10:
Another duplicate of https://github.com/FreeCAD/FreeCAD/issues/15102.

maxwxyz, 2024-07-10:
Duplicate of #15102

closed 2024-07-10 by maxwxyz