FreeCAD/FreeCAD #15342: Unclear/uncaught error: <class 'ValueError'>: Datum 7,09 mm for the constraint with index 20 is invalid

amreo, 2024-07-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

**Summary:**
I get the uncaught error when applying a vertical distance constrain to a vertical line. I expected to simply put the vertical distance constrain or at least to view a meaningful and caught error message.
```
12:42:41  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Datum 7,09 mm for the constraint with index 20 is invalid
12:42:41  App.getDocument('Unnamed').getObject('Sketch').setDatum(20,App.Units.Quantity('7.090000 mm'))
```
**Step to reproduce**
1) open this file --> [sample.FCStd.zip](https://github.com/user-attachments/files/16159660/sample.FCStd.zip)
2) go to part design workbench and enter in edit mode of `Sketch`
3) click to Edge1 (the most left vertical line)
4) press the I key (or to the constrain vertical distance button)
5) Write 5
6) Press enter
Now in the report should show a error like 
```
12:53:04  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Datum 5,00 mm for the constraint with index 27 is invalid
12:53:04  App.getDocument('Unnamed').getObject('Sketch').setDatum(27,App.Units.Quantity('5.000000 mm'))
12:53:04  The Sketch has partially redundant constraints!
```

**How this bug impact user experience**
This bug is unclear and I don't managed to understand the cause of this error/what I done wrong so this bug degrade the UX.

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * fasteners 0.5.0
  * 3DfindIT 1.2.0
  * freecad.gears 1.0.0
  * LCInterlocking
  * ModernUI (Disabled)
  * Beltrami 1.0.8-alpha
  * CurvedShapes 1.0.5
  * boltsfc 2022.11.5
  * ThreadProfile 1.89.0
  * sheetmetal 0.4.0
  * lattice2 1.0.0
  * Curves 0.6.22
  * Assembly4 0.50.6
  * Assembly3 0.12.2
  * A2plus 0.4.63
[/code]
```

### Subproject(s) affected?

PartDesign

### Anything else?

* [sample.FCStd.zip](https://github.com/user-attachments/files/16159660/sample.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

krushia, 2024-07-10:
In this case Sketcher is preventing creation of new constraints because of a preexisting "redundant constraints" condition. You need to first remove any redundant constraints. Alternatively you can click the checkbox to make it a reference constraint.

I agree the error text is misleading (gives the impression that something is corrupt).

Even more annoying is encountering this error when creating a dimensional constraint with a value that isn't possible based on other constraints. Especially after spending lots of time crafting an expression for it... only to see it go *poof* when clicking OK.

[image]

krushia, 2024-07-10:
Possibly duplicate of https://github.com/FreeCAD/FreeCAD/issues/5994 (a very old bug).

krushia, 2024-07-10:
Before maxwxyz asks:
Also present in dev version

OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * too many

maxwxyz, 2024-10-20:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/5994

closed 2024-10-20 by maxwxyz