FreeCAD/FreeCAD #15344: Assembly: exploded views do not work well with non-single-solid sub-assemblies

furgo16, 2024-07-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm trying to create an exploded view of a model which contains sub-assemblies. The sub-assemblies should also be exploded. The attached file is a very simplified version of the model I'm working with.

I encountered many strange issues when trying to do this, some of which I cannot easily reproduce (parts disappearing, focus jumping from one document to another). Here's a video of the ones I could reproduce.

Notice, when the checkbox `Parts as single solid` is unchecked on the Exploded view's Tasks view:

1. Even though only one sub-assembly is selected, the other one always moves along with it. Sub-assemblies cannot be independently moved
2. When exiting the Exploded Views object, the model needs to be manually recomputed to bring the components to their original location (I'm not sure if this is intended)
3. Moving a separate type of sub-assembly affects the position of other sub-assemblies. See at around 1:07 s in the video how moving the glazing bead component "pushes" the previously moved glass component. 

[AsParts.webm](https://github.com/FreeCAD/FreeCAD/assets/148809153/731245d1-a075-446e-85b0-ea9a0ba1e892)

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38095 (Git) Snap 1072
Build type: Release
Branch: main
Hash: af1f7385d47b9887e758794afdc3b1f69b406b96
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.7.3
  * Curves 0.6.40
  * ezydxf
  * Ondsel-Lens 2024.7.5.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-07-19:
So the problem here is that you are using links to assemblies. Because of that the objects within are not links but the external objects themselves.
So when one moves, the other moves as well because it's the same object.

If we want to make this work we need to fix the following issue as I describe in this comment https://github.com/FreeCAD/FreeCAD/issues/15208#issuecomment-2211150976

furgo16, 2024-07-23:
Related forum thread: https://forum.freecad.org/viewtopic.php?p=772142#p772142

furgo16, 2024-11-03:
Same behavior after merging https://github.com/FreeCAD/FreeCAD/pull/15629

PaddleStroke, 2024-11-04:
Have you replaced the App::Links with AssemblyLinks?

closed 2024-11-09 by chennes