FreeCAD/FreeCAD #15345: Repeated clicks on the 3D window no longer working for selecting up the tree view

furgo16, 2024-07-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Forum thread: https://forum.freecad.org/viewtopic.php?p=770471#p770471

Using Part Design, on a part containing a body with features it should be posssible triple click to start selecting a feature on the 3D view and work your way up to select the Part component, as described on https://wiki.freecad.org/Selection_methods#Repeated_clicks

I don't seem to be able to do this anymore on the weekly builds. The first click on the component on the 3D view selects the feature, but subsequent clicks do not select its parent containers. 

Steps to reproduce:

1. Open attached file
2. Click on a face of the component in the 3D window => PASS (it selects the last feature tip)
3. Click again => FAIL (nothing happens)
4. Click again => FAIL (nothing happens)

**Expected**:

https://github.com/FreeCAD/FreeCAD/assets/148809153/ec6dc05f-a62d-4b01-98b7-1a56a3b17699

**Actual**:

Repeated clicking on a component on the 3D window has no effect other than the first click, which selects the tip feature only.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38095 (Git) Snap 1072
Build type: Release
Branch: main
Hash: af1f7385d47b9887e758794afdc3b1f69b406b96
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.7.3
  * Curves 0.6.40
  * ezydxf
  * Ondsel-Lens 2024.7.5.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Core

### Anything else?

Quoting @Syres916 (thanks for testing!):

> The working build I have is:
>
```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git) AppImage
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * SheetMetal 0.4.15
  * Silk 0.1.5
  * FreeCAD-themes 2024.7.24
  * Curves 0.6.40
  * Help 1.0.3
  * CfdOF 1.27.4
  * fasteners 0.5.24
``` 
>
> So it's a regression since mid May 2024 especially as the Wiki still states the functionality unless someone's forgotten to update it!!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-10:
Does it interfere with the sync selection settings?

furgo16, 2024-07-10:
I'm not sure. Here are my settings:

[image]

It's really odd though. I've tried the steps to reproduce again, with the file in the description. Repeated clicks to select now work as expected. I can't think of anything I've changed since then, but it was definitely not working yesterday.

I'll leave it open for now for someone else to confirm, and unless they can, the issue can be closed.

maxwxyz, 2024-07-10:
I cannot reproduce, works as expected with these settings:
[image]

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git)
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.40
  * dodo 1.0.1
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.22

```

Syres916, 2024-07-10:
So as I predicted it's down to the file that an upgrading user from a pre-TNP build opened in a TNP build. the Python output makes it obvious there's a bug. Using build 37302 the three clicks produce the following output:

```
Gui.runCommand('Std_DrawStyle',6)
Gui.Selection.addSelection('SquareToRoundPipe','Part','Pipe.Edge28',1.00897e-10,-325,17.0735)
Gui.Selection.clearSelection()
Gui.Selection.addSelection('SquareToRoundPipe','Part','Pipe.',1.00897e-10,-325,17.0735)
Gui.Selection.clearSelection()
Gui.Selection.addSelection('SquareToRoundPipe','Part','',1.00897e-10,-325,17.0735)
```

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git) AppImage
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * SheetMetal 0.4.15
  * Silk 0.1.5
  * FreeCAD-themes 2024.7.24
  * Curves 0.6.40
  * Help 1.0.3
  * CfdOF 1.27.4
  * fasteners 0.5.24
```

with the same file and the same config files, build 38079 produces this output in the Python Console:

```
Gui.runCommand('Std_DrawStyle',6)
Gui.Selection.addSelection('SquareToRoundPipe','Part','Pipe.Edge28',1.00897e-10,-325,17.241)
Gui.Selection.clearSelection()
Gui.Selection.addSelection('SquareToRoundPipe','Part','Pipe.Edge28',1.00897e-10,-325,17.241)
Gui.Selection.clearSelection()
Gui.Selection.addSelection('SquareToRoundPipe','Part','Pipe.Edge28',1.00897e-10,-325,17.241)
```

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git) AppImage
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * SheetMetal 0.4.15
  * Silk 0.1.5
  * FreeCAD-themes 2024.7.24
  * Curves 0.6.40
  * Help 1.0.3
  * CfdOF 1.27.4
  * fasteners 0.5.24

```

It's an educated guess but not proven that it's linked to #15145 especially when you read https://github.com/FreeCAD/FreeCAD/issues/15145#issuecomment-2211434970

@furgo16 you weren't dreaming!!

furgo16, 2024-07-10:
> @furgo16 you weren't dreaming!!

@Syres916 thank you! :-)

maxwxyz, 2024-07-10:
@bgbsww FYI

furgo16, 2024-07-10:
I've updated the steps to reproduce. What seems to trigger it is opening a file created with the stable version (0.21) with the dev version and then doing a recompute. Before the recompute, you can still use repeated clicks, but not after.

bgbsww, 2024-07-10:
Interesting.  Thanks for the deep sleuthing.  Let's see if we can figure this one out.

So a pre-TNP file won't have an elementMap, so any looking up mappedNames is not going to hit.  After a recompute you're going to have elementMaps.  So we're looking for code related to selection that is either changed or failed to update to acknowledge mappedNames.  That is very likely to be the things I found on the way to the #15145 solution that weren't needed in the end.  SoFCUnifiedSelection.cpp is the leading candidate here based on all of our hunches.

bgbsww, 2024-07-10:
Scratch that.  SoFCUnifiedSelection::setSelection is useful to trace from, but it is really the question of what the difference is between:

1) Unrecomputed 0.21.1 file geometry (good)
2) Recomputed 021.1 file geometry (bad)
3) New drawn geometry (good)

Recompute calls all the execute() methods, and something isn't right there.

Really fascinating looking at a file where one object works right, and the one next to it does not.

If you select the Body in the tree, and then click again on a face in the 3d view it correctly steps up a notch to selecting the Part.  Click again and it loops down to the face, at which point it gets stuck again.

Bad:
>>> App.ActiveDocument.Pad.Shape.Faces[3].ElementMap
{'g5;SKT;:H12dc,E;:G;XTR;:H12dc:7,F;:U;XTR;:H12dc:7,E;:U2;XTR;:H12dc:8,V': 'Vertex2', 'g5v2;SKT;:H12dc,V': 'Vertex1', 'g5v2;SKT;:H12dc,V;:G;XTR;:H12dc:7,E': 'Edge1', 'g6;SKT;:H12dc,E': 'Edge3', 'g6;SKT;:H12dc,E;:G;XTR;:H12dc:7,F': 'Face1', 'g6;SKT;:H12dc,E;:G;XTR;:H12dc:7,F;:U;XTR;:H12dc:7,E': 'Edge4', 'g6;SKT;:H12dc,E;:G;XTR;:H12dc:7,F;:U;XTR;:H12dc:7,E;:U2;XTR;:H12dc:8,V': 'Vertex4', 'g6v1;SKT;:H12dc,V': 'Vertex1', 'g6v2;SKT;:H12dc,V': 'Vertex3', 'g6v2;SKT;:H12dc,V;:G;XTR;:H12dc:7,E': 'Edge2', 'g7v1;SKT;:H12dc,V': 'Vertex3'}
>>> App.ActiveDocument.Pad.Shape.ElementMap
{'g3;SKT;:H12dc,E': 'Edge3', 'g3;SKT;:H12dc,E;:G;XTR;:H12dc:7,F': 'Face1', 'g3;SKT;:H12dc,E;:G;XTR;:H12dc:7,F;:U;XTR;:H12dc:7,E': 'Edge4', 'g3;SKT;:H12dc,E;:G;XTR;:H12dc:7,F;:U;XTR;:H12dc:7,E;:U2;XTR;:H12dc:8,V': 'Vertex4', 'g3;SKT;:H12dc,E;:G;XTR;:H12dc:7,F;:U;XTR;:H12dc:7,E;:U;XTR;:H12dc:7,V': 'Vertex2', 'g3;SKT;:H12dc,E;FAC;:H12dc:4,F': 'Face8', 'g3;SKT;:H12dc,E;FAC;:H12dc:4,F;:G0;XTR;:H12dc:8,F': 'Face9', 'g3v1;SKT;:H12dc,V': 'Vertex1', 'g3v1;SKT;:H12dc,V;:G;XTR;:H12dc:7,E': 'Edge1', 'g3v2;SKT;:H12dc,V': 'Vertex3', 'g3v2;SKT;:H12dc,V;:G;XTR;:H12dc:7,E': 'Edge2', 'g4;SKT;:H12dc,E': 'Edge6', 'g4;SKT;:H12dc,E;:G;XTR;:H12dc:7,F': 'Face2', 'g4;SKT;:H12dc,E;:G;XTR;:H12dc:7,F;:U;XTR;:H12dc:7,E': 'Edge7', 'g4;SKT;:H12dc,E;:G;XTR;:H12dc:7,F;:U;XTR;:H12dc:7,E;:U2;XTR;:H12dc:8,V': 'Vertex6', 'g4v1;SKT;:H12dc,V': 'Vertex3', 'g4v2;SKT;:H12dc,V': 'Vertex5', 'g4v2;SKT;:H12dc,V;:G;XTR;:H12dc:7,E': 'Edge5', 'g5;SKT;:H12dc,E': 'Edge9', 'g5;SKT;:H12dc,E;:G;XTR;:H12dc:7,F': 'Face3', 'g5;SKT;:H12dc,E;:G;XTR;:H12dc:7,F;:U;XTR;:H12dc:7,E': 'Edge10', 'g5;SKT;:H12dc,E;:G;XTR;:H12dc:7,F;:U;XTR;:H12dc:7,E;:U2;XTR;:H12dc:8,V': 'Vertex8', 'g5v1;SKT;:H12dc,V': 'Vertex5', 'g5v2;SKT;:H12dc,V': 'Vertex7', 'g5v2;SKT;:H12dc,V;:G;XTR;:H12dc:7,E': 'Edge8', 'g6;SKT;:H12dc,E': 'Edge12', 'g6;SKT;:H12dc,E;:G;XTR;:H12dc:7,F': 'Face4', 'g6;SKT;:H12dc,E;:G;XTR;:H12dc:7,F;:U;XTR;:H12dc:7,E': 'Edge13', 'g6;SKT;:H12dc,E;:G;XTR;:H12dc:7,F;:U;XTR;:H12dc:7,E;:U2;XTR;:H12dc:8,V': 'Vertex10', 'g6v1;SKT;:H12dc,V': 'Vertex7', 'g6v2;SKT;:H12dc,V': 'Vertex9', 'g6v2;SKT;:H12dc,V;:G;XTR;:H12dc:7,E': 'Edge11', 'g7;SKT;:H12dc,E': 'Edge15', 'g7;SKT;:H12dc,E;:G;XTR;:H12dc:7,F': 'Face5', 'g7;SKT;:H12dc,E;:G;XTR;:H12dc:7,F;:U;XTR;:H12dc:7,E': 'Edge16', 'g7;SKT;:H12dc,E;:G;XTR;:H12dc:7,F;:U;XTR;:H12dc:7,E;:U2;XTR;:H12dc:8,V': 'Vertex12', 'g7v1;SKT;:H12dc,V': 'Vertex9', 'g7v2;SKT;:H12dc,V': 'Vertex11', 'g7v2;SKT;:H12dc,V;:G;XTR;:H12dc:7,E': 'Edge14', 'g8;SKT;:H12dc,E': 'Edge18', 'g8;SKT;:H12dc,E;:G;XTR;:H12dc:7,F': 'Face6', 'g8;SKT;:H12dc,E;:G;XTR;:H12dc:7,F;:U;XTR;:H12dc:7,E': 'Edge19', 'g8;SKT;:H12dc,E;:G;XTR;:H12dc:7,F;:U;XTR;:H12dc:7,E;:U2;XTR;:H12dc:8,V': 'Vertex14', 'g8v1;SKT;:H12dc,V': 'Vertex11', 'g8v2;SKT;:H12dc,V': 'Vertex13', 'g8v2;SKT;:H12dc,V;:G;XTR;:H12dc:7,E': 'Edge17', 'g9;SKT;:H12dc,E': 'Edge20', 'g9;SKT;:H12dc,E;:G;XTR;:H12dc:7,F': 'Face7', 'g9;SKT;:H12dc,E;:G;XTR;:H12dc:7,F;:U;XTR;:H12dc:7,E': 'Edge21', 'g9v1;SKT;:H12dc,V': 'Vertex13', 'g9v2;SKT;:H12dc,V': 'Vertex1'}

Good:

>>> App.ActiveDocument.Box.Shape.Faces[5].ElementMap
{'Edge10': 'Edge4', 'Edge12': 'Edge2', 'Edge2': 'Edge1', 'Edge6': 'Edge3', 'Face6': 'Face1', 'Vertex1': 'Vertex2', 'Vertex3': 'Vertex1', 'Vertex5': 'Vertex4', 'Vertex7': 'Vertex3'}
>>> App.ActiveDocument.Box.Shape.ElementMap
{}
>>>

furgo16, 2024-07-10:
@bgbsww it's beyong my skills to be able to follow this, but I really appreciate the great work you're doing chasing and fixing these TNP bugs. Thank you, and keep up the good work!

bgbsww, 2024-07-10:
Thanks, we're all doing our piece here - good bug reports are gold, because I couldn't necessarily have done all that work any faster than anyone else.  They let me focus on the fixes.

In this case, we seem to have a situation where the subshape elementMap for Face6 of the box I drew (  Faces[5] ) has only simple classic indexed names in it: "Edge4" and the box itself has an empty element map.  That works.  Wheareas the Part read in from the older file and recomputed has full mapped names in the element Map.

PR #15381 (2024-07-15): Toponaming: Process mapped names during selection
Fix #15345.
closed 2024-07-15 by chennes