FreeCAD/FreeCAD #15346: Classic Theme: Qt6 on Windows does not adapt appropriately

chennes, 2024-07-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When compiled against Qt6 and run on Windows 11, the interaction of FreeCAD's widgets with Windows icons and fonts results in a sometimes-unusable (and always inconsistent) user interface. In particular, if Windows 11 is running in dark mode the menu text remains black, which cannot be seen against the Win11 dark mode menu background. Even in light mode, the default system icons give an inconsistent look to the interface and should not be used.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37953 +8 (Git)
Build type: Release
Branch:  main
Hash: c29cb8464d02469f21a53b34d8791cd6941fe722
Python 3.12.3, Qt 6.7.1, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

LarryWoestman, 2024-07-10:
I did a test build of the chennes' new libpack 3 release candidate:

`
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38095 (Git)
Build type: Release
Branch: main
Hash: af1f7385d47b9887e758794afdc3b1f69b406b96
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: English/United States (en_US)
`
It appears to me that it occurs on many dialogs, but not always consistently.  Sometimes the dialogs are completely dark on dark and other times the dialogs contain some light on dark entries.  Three representative screen shots:

[image]
[image]
[image]

MisterMakerNL, 2024-07-20:
We have set QT darkmode to 1 but this should only affect the window decoration like title bar and borders. This worked perfectly fine in QT 5, but stopped working in QT 6.
https://github.com/FreeCAD/FreeCAD/blob/608cb182e516e683befa890a84ca615c040b1235/src/Main/MainGui.cpp#L161

luzpaz, 2024-08-24:
@kadet1090 :point_up_2:

luzpaz, 2024-10-23:
Moved to Post 1.0 milestone where Qt6 issues will be addressed (unless someone wants to spearhead this one beforehand, please have at it)

bosmak, 2025-04-03:
I'm on QT6, and it seems that just passing the argument `-platform windows:darkmode=2` made it work, but unfortunately, it won't work since FreeCAD says this is an unrecognized option and doesn't launch the program

[image]

Note that if I launch it with darkmode=1, the messagebox with the error is light themed.

[image]

As a workaround, adding the following lines to the qt6.conf in the bin folder made it work

```
[Platforms]
WindowsArguments = darkmode=2
```

Maybe changing the value from 1 to 2 in the line @MisterMakerNL mentioned will fix this issue.

MisterMakerNL, 2025-04-03:
If I change darkmode too 2 here: https://github.com/MisterMakerNL/FreeCAD_MisterMaker/blob/db08735e106561f3a11cf08e02fbf5d9e8fbf6b2/src/Main/MainGui.cpp#L184
It does work.
But also applies a terrible stylesheet too the no-stylesheet option:

[image]

MisterMakerNL, 2025-04-03:
Found some more info here:
https://bugreports.qt.io/browse/QTBUG-112975

> If darkmode support is set to 1, then the window should have a dark frame if it has a dark palette. If it doesn't have a dark palette, the frame should be light (ie. the frame follows the style of the UI).

> With darkmode set to 2, the default palette will come from the system, respecting a dark palette for a system running in dark mode. However, a native style that doesn't support dark mode (like the native Windows style, which cannot support dark mode) will override that with a light palette no matter the "darkmode" parameter.

> Window, not being a Qt Quick Control element, has no style, so it will always return the system palette, unmodified by any style. So I assume you get the dark frame with Window, but the light frame with ApplicationWindow, unless you explicitly use a Qt Quick style that supports a dark palette, like Fusion.

So basically to fix this, we have to check if the stylesheet is dark, then set a dark palette... Which is kinda dumb in my opinion the older method worked fine.

luzpaz, 2026-09-22:
This is probably obsolete, right @chennes  ?