FreeCAD/FreeCAD #15351: Objects disapear when in Sketch mode

leoheck, 2024-07-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Title says it all. The only thing I can see is like a projection on the 2d plane of the Current Body I am modeling.
It does not show the Subshapebinders, it does not show Parts.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git) AppImage
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.22
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * asm4_disable_version_check_popup
  * OpticsWorkbench 1.0.18
  * lattice2 1.0.0
  * kicadStepUpMod
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-07-10:
What do you have set for **Preferences > Sketcher > Display > Hide all object that depend on the sketch**?

maxwxyz, 2024-07-11:
Cannot reproduce on Windows

leoheck, 2024-07-11:
When I go to the Sketch Edition, objects disappear and I cannot toggle then visible again while in the Sketch Edition. For instance, I would like to copy external features from a Subshapebinder or Sketch based on existing objects. If I move the view a bit at some specific 3d angles, I can see the the objects that have disappeared, but then I am not looking from over the face I am editing.

leoheck, 2024-07-11:
The whole Freecad is freezing for 1 or 2 seconds eventually, periodically.

I am seeing this, from ZX
[image]

And this from XY (look the details)
[image]

When I go to the Sketch Edition (Sketch on the XY_Plane), things have disappeared

the ZX looks like this
[image]

the XY looks like this.
[image]

If I move the view a bit, then I start seeing another bit of it.
[image]

> What do you have set for **Preferences > Sketcher > Display > Hide all object that depend on the sketch**?

@depthoffocus 

[image]

leoheck, 2024-07-11:
@depthoffocus I didn't check this settings, it is the default thing.

depthoffocus, 2024-07-11:
> @depthoffocus I didn't check this settings, it is the default thing.

The behaviour you're talking about sounds like it is the expected behaviour given that tickbox. 

The default is the old-fashioned notion of hiding the dependent objects so the sketch is easier to see. You could try un-ticking that box to see if it provides a behaviour you expect. This is covered in this video:

https://www.youtube.com/watch?v=0L0mVEcMbaM

leoheck, 2024-07-11:
I never edited Sketch settings, this started on the latest dev release. I cannot even see the Subshape binders... 
There is also something wrong happening with transparency... going to the Sketch Edit disables transparency or adding features makes parts opaque. Something is not right with this version.

depthoffocus, 2024-07-11:
> There is also something wrong happening with transparency... going to the Sketch Edit disables transparency or adding features makes parts opaque. Something is not right with this version.

I gather this is a workaround feature designed to make it easier to deal with sketches when the scene illumination would render the object behind it either plain white or plain black due to shadow/reflectance, which can be maddening.

It too can be switched off: **Preferences > Sketcher > General > Disable shading in edit mode**

depthoffocus, 2024-07-11:
(Deleted previous comment -- misunderstood you)

depthoffocus, 2024-07-11:
@leoheck oddly enough I am now getting this -- I have a sketch I want to copy geometry from and I can't get it to appear long enough on the screen to do so. It flickers as I hover over it in the tree view. I'm going to try to extract a reproducible case.

leoheck, 2024-07-11:
It is somehow showing only things under the Sketch considering the angle of the view we are looking at it.

depthoffocus, 2024-07-11:
> It is somehow showing only things under the Sketch considering the angle of the view we are looking at it.

I am not sure if what I am seeing is the expected behaviour. I can still copy the geometry I want if I switch out of "section" view but I'm getting a crash from the geometry and I'm not sure why; I could be introducing a dependency cycle somewhere, I must try to simplify the problem.

leoheck, 2024-07-11:
can you share your project? or a sample that has the objects that are crashing?

depthoffocus, 2024-07-11:
I am going to try to isolate the elements that crash; I think I have a
reliable way to get there but I am trying to simplify it as much as possible

luzpaz, 2024-07-27:
@depthoffocus any progress ?

depthoffocus, 2024-07-27:
> @depthoffocus any progress ?

My crash was caused by #15154, which is fixed.

luzpaz, 2024-09-03:
bumping for visibility

leoheck, 2025-02-24:
Nope.

closed 2025-02-24 by leoheck