FreeCAD/FreeCAD #15352: Sketcher disabled shading sometimes persists after recomputing in Sketch view

depthoffocus, 2024-07-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

**Preferences > Sketcher > Display > Hide all objects that depend on the sketch** is disabled

**Preferences > Sketcher > General > Disable shading in edit mode** is enabled

If you:
- go in to the Sketch for the extrude
- make a change
- force the recompute of that change so it updates

then when you exit Sketcher the disabled shading view persists.

It will only change if the tree is changed (e.g. deleting the Fusion). 

This does not happen if you exit **without** recomputing.

Hope my animated gif is useful

[image]

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git)
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.90.0
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * Curves 0.6.40
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-07-13:
This issue seems to be resolved (at least in my test case above) as of the most recent weekly:

OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38127 (Git)
Build type: Release
Branch: main
Hash: 9868533d8d0ef1f569cbde538b7a31350db503bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.90.0
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * Curves 0.6.40

luzpaz, 2024-07-13:
Is there a commit associated with this fix ?

depthoffocus, 2024-07-14:
> Is there a commit associated with this fix ?

Sorry, ignore my previous message. It's not resolved, in fact; I somehow confused myself. My apologies.

To add another fact to the picture: you don't need to make a change at all to trigger the bug. 

It is triggered by simply forcing a recompute.

depthoffocus, 2024-08-01:
As far as I can see, forcing a recompute no longer generates this problem.

But with the Sketcher preferences set to not hide the dependent objects:

[image]

The following steps cause the problem:

- draw any simple sketch
- extrude it
- go back into the sketch
- render it un-closed by adding geometry
- remove that geometry
- leave sketcher

The disable-shading style will be stuck on (changing the height of of the extrude will fix it)

https://github.com/user-attachments/assets/1bfabcf7-6a0e-4c97-80ee-492423f9b520

OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38375 (Git)
Build type: Release
Branch: main
Hash: fe702349e848571802c01659668cc4bee519f4c6
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.90.0
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * Curves 0.6.42

nexx512, 2024-11-22:
I experienced the same issue but independent of the "hide dependent objects" settings. However, when "hide dependent objects" is disabled, the shading is wrong right when entering the sketcher view.

**Used Freecad Version**
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1202
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly4 0.50.15
  * fasteners 0.5.30
  * sheetmetal 0.5.5

**Example with "hide dependent object" enabled:**
[Bildschirmaufzeichnung vom 2024-11-22 17-45-49.webm](https://github.com/user-attachments/assets/cf63bdb2-fe1e-4d01-a5ef-94c20ef1bf83)

**Example with "hide dependent object" disabled:**
[Bildschirmaufzeichnung vom 2024-11-22 18-11-49.webm](https://github.com/user-attachments/assets/bb98dfe4-ae76-47fe-bf5e-422ddf15ebb0)

closed 2025-02-24 by chennes