FreeCAD/FreeCAD #15353: PartDesign hold: "Removing splitter failed" with multiple holes or when threaded

theosib, 2024-07-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a super simple design here. Nothing fancy or complicated. Please see the attachment.

I'm trying to add some M3 threaded screw holes. So I have attached a sketch to a surface and have positioned four circles. Back in PartDesign, I select that sketch and create holes for them.

If the holes are not threaded, they work fine.

If there is only one hole (I turn three of them into construction geometry), and I enable threading, then it works fine.

But if I enable more than one, then the hole creation fails with "Removing splitter failed." And of course, this is an entirely informative error message, so there's no hint as to how to fix it.
[screen_bracket.FCStd.zip](https://github.com/user-attachments/files/16168139/screen_bracket.FCStd.zip)

### Full version info

```shell
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-07-10:
I loaded up my design in 0.21.2 and made the holes the way I wanted to, where I have a single sketch with multiple holes.

It all works just fine. So this is definitely a regression.

theosib, 2024-07-11:
Incidentally, someone on Discord told me to turn off the "refine model" settings, because that helps sometimes with these screw hole problems. I have not tried this yet in a 0.22 version. However, FYI, my current settings in 0.21.2 that work are as shown in the image.
[image]

maxwxyz, 2024-07-11:
Please test if it works in 0.22 with the same settings.

theosib, 2024-07-11:
I just tried it in 38079. I get the same "Removing splitter failed" error. Changing the "refine" setting doesn't help.

theosib, 2024-08-02:
Just out of curiosity, what is the workaround?

luzpaz, 2024-08-02:
https://github.com/FreeCAD/FreeCAD/issues/15353#issuecomment-2223407106

drew585, 2024-08-09:
I'm having same issue:

OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git) AppImage
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.5.25

[image]

FC 21.3:
[image]

theosib, 2024-08-10:
> [#15353 (comment)](https://github.com/FreeCAD/FreeCAD/issues/15353#issuecomment-2223407106)

I'm not sure that falling back to 0.21.2 counts as a workaround. Moreover, while I can design it in 0.21.2, if I take that design and load it into 0.22dev, it immediately breaks.

laurencei, 2024-08-18:
I have the same issue. My screenshots are similar to the ones above - I am following the tutorial by MangoJelly, so I'm guessing some people above are as well. 

The tutorial is here: https://www.youtube.com/watch?v=KzLWKKBtnxU

The issue occurs at 30:57. Watching 27:40 onwards is the steps to reproduce.

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git)
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)

krushia, 2024-08-23:
FYI, this error only happens when two threaded holes on **_opposite corners_** are done in one sketch.

AIRCAP, 2024-11-12:
I can no longer reproduce the issue in latest main - this might have been fixed by the recent bugfixes on partdesign holes and/or boolean fuzzies. @theosib can you re-test with current main and confirm it works now?

maxwxyz, 2024-11-12:
Should be fixed in  https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Feel free to reopen if there is still an issue

naddeoa, 2024-12-05:
This still happens in that build for me. 

[image]

If you add threads onto the "Hole" step then that error pops up and the subsequent pad breaks. Here is a link to the project file: 

https://github.com/naddeoa/forum-files/blob/master/vent.FCStd

FlachyJoe, 2024-12-05:
"Removing splitter failed" is a refine error so switch off the feature Refine property and autorefine setting should fix the model.
Related: #17846

dwmw2, 2025-01-23:
I see the same problem with the latest weekly build (https://github.com/FreeCAD/FreeCAD/commit/a977fade2de741c94c9efa59b1c2c26df18eb631), and turning the refinement settings off as shown in https://github.com/FreeCAD/FreeCAD/issues/15353#issuecomment-2223407106 doesn't help.

With http://david.woodhou.se/BlindMount.FCStd if I set `VarSet.GrooveSize` to 5.5mm it gives me the 'Removing splitter failed' message. With 5.0 or 6.0mm it's fine.

[image]

ovo-Tim, 2025-02-01:
I also encounter this error when I build my workbench. Not sure if it's a bug or my wrong. But it seems it's not very easy to build a minimum example to recurrence since I used `PartDesign::FeaturePython`.

ovo-Tim, 2025-02-04:
I have tested the latest development version, and the issue persists. As mentioned previously, I noticed that by setting 'Refine' to False, the error disappears; however, this results in giving me the object that should be removed.

ovo-Tim, 2025-02-04:
> I also encountered this error while building my workbench. The correct shape still can be displayed, but it can't be recomputed. Not sure if it's a bug or my mistake. However, the minimum example I built seems can't recur the issue.

It turns out that the issue stems from the fact that the sketch generated by script didn't add constraints.

ovo-Tim, 2025-02-06:
> I have tested the latest development version, and the issue persists. As mentioned previously, I noticed that by setting 'Refine' to False, the error disappears; however, this results in giving me the object that should be removed.

It turns out that this was my mistake rather than a bug.

byte666, 2025-05-22:
I've never had a problem like this before. Since upgrading from 1.0 to 1.0.1, I get this error message when the following scenario occurs: Threaded hole, display thread, countersink, refine=true. If I set refine to false OR remove countersink OR remove display thread, the error disappears.

Panda631, 2025-05-23:
I can second this problem. During daily work I'm spending a lot of time to work around this problem.

FlachyJoe, 2025-05-23:
Simply disable the auto-refine preference and set the Refine property only when you really need it.

Panda631, 2025-05-23:
Thanks. I found that hint in the forum.
However it's a bug and most user didn't know this work around.

luzpaz, 2026-10-08:
This needs to be retested for 26.3 RC1 and latest dev