FreeCAD/FreeCAD #15355: CAM_ Post B axis error in generating g code

wtfiAndrewNelson, 2024-07-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After generating path information for my surfaces around the b axis (y axis) which looks correct, the post command generates negative values of gcode to a limit of -90 then reverses counting up to a limit of 90. I have tried multiple models and all of the post files that I could find. They all do the same thing.

Working with the A axis, everything is fine. But that doesn't help me as my rotational axis is in the Y orientation not the X.

I have tried all that I can come up with but to no avail. I am attaching screenshots from freecad and from UGS Platform, but I have also tested the gcode on my machine and it just rocks back and forth without going more than 180 degrees of rotation, -90 to 90 degrees.

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
[/code]
```

### Subproject(s) affected?

CAM/Path

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-11:
could you please test if the bug is also present in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
A demo file would also be great!

wtfiAndrewNelson, 2024-07-11:
I just tried it on the "FreeCAD_weekly-builds-33458-2023-06-28-conda-macOS-arm64-py310.dmg" and had the same outcome. Here are my files, both the design and the gcode.

[1 milling designs.zip](https://github.com/user-attachments/files/16182740/1.milling.designs.zip)

mosfet80, 2025-05-18:
I think it's a clipper problem.
try this patch https://github.com/FreeCAD/FreeCAD/pull/21437

FEA-eng, 2025-10-01:
@wtfiAndrewNelson Can you confirm if it's still relevant?

LarryWoestman, 2025-10-22:
I'm glad to hear it!

Thanks for letting us know.