FreeCAD/FreeCAD #15361: Assembly view provider doesn't support the transform manipulator

wwmayer, 2024-07-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

* Create an assembly object or load a project with an assembly
* Double-click the assembly to deactivate it
* Select the assembly and open the context-menu
* Click on the Transform command
* The transform manipulator to change the placement doesn't appear
* Instead the assembly has become active now

The reason is the implementation of ViewProviderAssembly::setEdit https://github.com/FreeCAD/FreeCAD/blob/342116e61480fa70bd94454a94095ed4d1db940a/src/Mod/Assembly/Gui/ViewProviderAssembly.cpp#L183 because it completely ignores the edit mode.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38112 (Git)
Build type: debug
Branch: main
Hash: fb6c84b717765066a4ba796c1b1c09f38702fd43
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-07-15:
> because it completely ignores the edit mode.

Can you please elaborate?

The transform tool is indeed not working on assembly objects. If I have to guess, this transform tool is actually not triggering the transform operation, but instead it is setting edit on the object. Which for App::Part for example triggers the transform dialog.

Any fix suggestion?

wwmayer, 2024-07-15:
Usually setEdit/unsetEdit are implemented this way:
```
bool ViewProviderAssembly::setEdit(int mode)
{
    if (mode == ViewProvider::Default) {
... // edit mode of this sub-class
     return true;
    }

    return ViewProviderPart::setEdit(mode);
}
```
unsetEdit must be implemented accordingly.

closed 2024-07-17 by chennes