FreeCAD/FreeCAD #15363: BIM: NativeIFC file with basic structure does not have level on reopening

Roy-043, 2024-07-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The basic structure of a "Strict IFC" document contains 3 levels: Site > Building > Level. The level is missing when a file with that structure is saved and reopened.

Steps:
1. Start FreeCAD.
2. Switch to the BIM WB.
3. Switch ON "strict IFC" in the statusbar.
4. Answer "Yes" in the popup to create a default structure.
5. Save with the "Save IFC file as..." menu option.
6. Reopen the file.
7. Result: Level is missing.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git)
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-07-11:
The level is there, it's simply not shown by default when opening an IFC file. You need to double-click the building. Unless you change the NativeIFC preferences to expand building structure on load

Roy-043, 2024-07-11:
Ah yes, you are right, the Level is there. But double-clicking the Building does not expand the node if it contains a single empty Level. You have to add something to the Building for the node to become expandable.

yorikvanhavre, 2024-07-15:
Indeed, that's a bug. I'll submit a fix

PR #15458 (2024-07-16): BIM: Fixed empty IFC decomposition
Fixes #15363
commit 4d29b50e (2024-07-16): BIM: Fixed empty IFC decomposition - fixes #15363
closed 2024-07-16 by yorikvanhavre