FreeCAD/FreeCAD #15365: Material: What should happen to the "Override colors" option of Links?

Roy-043, 2024-07-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Links have an "Override colors" option. The new material system has not yet been introduced there. Is this planned for the V1.0 release?

### Full version info

```shell
V0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-11:
@davesrocketshop is this feasible for 1.0?

davesrocketshop, 2024-07-11:
> @davesrocketshop is this feasible for 1.0?

Let me investigate. I have no idea what's involved

Roy-043, 2024-09-18:
Note that this option supports transparency through the alpha values of the OverrideColorList. But because of a regression since V0.20 this transparency no longer be set via the GUI. See #16637.

luzpaz, 2026-10-08:
Will a fix for this make it into 26.3 ?