FreeCAD/FreeCAD #15372: Links to nested assemblies have an oversized overlay icon

furgo16, 2024-07-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When adding nested assemblies to an assembly, the overlay icons on the tree view that indicate that they are a link are oversized and clipped in comparison to the same [overlay icons](https://wiki.freecad.org/Tree_view#Overlay_icons) from other components (compare e.g. Assembly001 with Muntin bar002 in the screenshot below) 

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38120 (Git) Snap 1073
Build type: Release
Branch: main
Hash: 9eeca1f6ad50cea9fa2d30d1a4fbb8b6e8fb4bbd
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.7.3
  * Curves 0.6.40
  * ezydxf
  * Ondsel-Lens 2024.7.5.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-07-11:
I chose Assembly as the affected Subproject, but I think this is more of a global issue. Observing this somewhere else, it seems the issue applies to links to external files only.

maxwxyz, 2024-07-11:
Related: https://github.com/FreeCAD/FreeCAD/issues/6765
On some systems they appear too small

furgo16, 2024-07-11:
This is marked as https://github.com/FreeCAD/FreeCAD/labels/Good%20first%20issue, but is the actual cause and the steps to fix it known?

Looking at the related issue, it seems that there are a number of xpm files that need to be removed from git and from a resource file, but simply for tidying up purposes. If I understand it correctly, their .svg counterparts are already being used, but the size is still an issue?

furgo16, 2024-10-09:
Can someone knowledgeable with this issue perhaps summarize what's needed to do to fix it? I'd try to give a go to a fix once I understand it better. Thanks!

furgo16, 2024-10-09:
I'm looking at how it was fixed in https://github.com/FreeCAD/FreeCAD/commit/a93997df64e1401a0e89ec180b6222ff57e87bc0

These are the remaining XPM bitmaps

```
find -name *.xpm
./src/Gui/Icons/button_right.xpm
./src/Gui/Icons/bulb.xpm
./src/Gui/Icons/button_up.xpm
./src/Gui/Icons/LinkArrayOverlay.xpm
./src/Gui/Icons/LinkSubElement.xpm
./src/Gui/Icons/LinkOverlay.xpm
./src/Gui/Icons/freecad.xpm
./src/Gui/Icons/button_left.xpm
./src/Gui/Icons/button_down.xpm
./src/Gui/Icons/LinkSubOverlay.xpm
```

I would think that the overlay in question is one of `LinkSubElement`, `LinkOverlay` or `LinkSubOverlay`. However, it seems their SVG counterparts are already being loaded:

```
git grep "LinkSubElement"
src/Gui/Icons/LinkSubElement.svg:   sodipodi:docname="LinkSubElement.svg"
src/Gui/Icons/LinkSubOverlay.svg:   sodipodi:docname="LinkSubElement.svg"
src/Gui/Icons/resource.qrc:        <file>LinkSubElement.svg</file>
src/Gui/Icons/resource.qrc:        <file>LinkSubElement.xpm</file>
src/Gui/ViewProviderLink.cpp:        return BitmapFactory().pixmapFromSvg("LinkSubElement", QSizeF(px,px));
```

```
git grep "LinkOverlay"
src/Gui/Icons/LinkArrayOverlay.svg:   sodipodi:docname="LinkOverlay.svg"
src/Gui/Icons/LinkOverlay.svg:   sodipodi:docname="LinkOverlay.svg"
src/Gui/Icons/resource.qrc:        <file>LinkOverlay.svg</file>
src/Gui/Icons/resource.qrc:        <file>LinkOverlay.xpm</file>
src/Gui/Tree.cpp:                pxExternal = Gui::BitmapFactory().pixmapFromSvg("LinkOverlay",
src/Gui/ViewProviderLink.cpp:        return BitmapFactory().pixmapFromSvg("LinkOverlay", QSizeF(px,px));
```

```
git grep "LinkSubOverlay"
src/Gui/Icons/resource.qrc:        <file>LinkSubOverlay.svg</file>
src/Gui/Icons/resource.qrc:        <file>LinkSubOverlay.xpm</file>
src/Gui/ViewProviderLink.cpp:        return BitmapFactory().pixmapFromSvg("LinkSubOverlay", QSizeF(px,px));
```

maxwxyz, 2024-10-09:
@furgo16 thank you! Would be great for 1.0!

furgo16, 2024-10-09:
WIP...

| XPM icon name | XPM icon | SVG icon name | SVG icon |
| --- | --- | --- | --- |
| button_right.xpm  | [image] | button_right.svg | [image] |
| bulb.xpm | [image] | bulb.svg | [image] |
| button_up.xpm | [image] | button_up.svg | [image] |
| LinkArrayOverlay.xpm | [image]  |  |  |
| LinkSubElement.xpm | [image]  |  |  |
| LinkOverlay.xpm |[image] | LinkOverlay.svg | [image] |
| freecad.xpm | [image] |  |  |
| button_left.xpm | [image] |  |  |
| button_down.xpm | [image] |  |  |
| LinkSubOverlay.xpm  | [image]  |  |  |

maxwxyz, 2024-10-09:
@furgo16 do you need some svgs? I can provide them but I think they should be all in GUI

furgo16, 2024-10-09:
Thanks! It should be fine, the SVGs seem to be all there, it's just that the table is WIP and I come back to it when I have a few spare minutes.

In any case, I think the XPM icons are not the cause. If I understand it correctly, there seems to be a difference in how the size of the overlay icons is calculated in local links vs external links:

https://github.com/FreeCAD/FreeCAD/blob/a9bff78974f301100134a13c636e23aa91be5bcf/src/Gui/ViewProviderLink.cpp#L1727-L1735

https://github.com/FreeCAD/FreeCAD/blob/a9bff78974f301100134a13c636e23aa91be5bcf/src/Gui/Tree.cpp#L5421-L5422

PR #17151 (2024-10-14): Resize external link overlays relative to the device's pixel ratio
Fixes: #15372

- Use the same resize method for local and external link overlay icons (before, external link overlay icons were hardcoded), based on the device pixel ratio (https://github.com/FreeCAD/FreeCAD/commit/a93997df64)
- Remove obsolete `src/Gui/Icons/LinkOverlay.xpm` icon to avoid confusion 
- In the table with screenshots below, `Cube002` is the external link to look at for the changes in the overlay icon. Notice the difference comparing the overlay icon with a local link (`Cube001`) before, and the same size after
-  [... cut]
closed 2024-10-14 by yorikvanhavre