FreeCAD/FreeCAD #15373: Assembly BOM considers groups as parts

furgo16, 2024-07-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using [groups](https://wiki.freecad.org/Std_Group) as a container to organize sets of components in an assembly, the BOM tool considers them as parts and lists them in the assembly. Even when choosing the `Only parts` checkbox in the BOM Tasks panel dialog, groups will be listed in the BOM.

Since there is already a distinction in FreeCAD between Groups and Parts, it would seem logical that the BOM is also consistent with that distinction. As it is right now, it's not possible to keep Groups out of the BOM output.

Groups might have a logical or organizational purpose as containers from components that have no functional relation, unlike parts, thus it would seem it does not make sense to make them behave as parts in the BOM. Additionally groups can be moderately deep-nested, which can add unnecessary clutter to the BOM.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38120 (Git) Snap 1073
Build type: Release
Branch: main
Hash: 9eeca1f6ad50cea9fa2d30d1a4fbb8b6e8fb4bbd
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.7.3
  * Curves 0.6.40
  * ezydxf
  * Ondsel-Lens 2024.7.5.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-07-11:
I agree. Groups should be purely visual. The BOM tool should consider their contents as direct children of the root objects the groups are inserted in.

PaddleStroke, 2024-07-12:
Got it, unless someone disagree and further discussion happens, I will make the change

tobiasfalk, 2024-07-15:
Yes, groups should not be considered as Parts and therefore not be a part in the BOM.
What would be nice if it would be possible to sort the BOM by the Groups or only get a BOM from one group.

closed 2024-07-22 by chennes