FreeCAD/FreeCAD #15377: Closing sketch with point causes segfault

oerkel47, 2024-07-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I added a point on an external geometry edge. Now every time I open this sketch and close it again FreeCad crashes. If I remove the point before closing the sketch it does not crash but a face is not found anymore. This project has an imported step file which is used a BaseFeature (I guess this is related to the problem).

FreeCAD file in question: [FreecadCrash.zip](https://github.com/user-attachments/files/16180854/FreecadCrash.zip)

https://github.com/FreeCAD/FreeCAD/assets/57231822/3ec1f02d-754b-4b1e-bb81-719c0a9cb795

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git) AppImage
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenTheme 2024.7.3
  * kicadStepUpMod 11.2.4
  * Manipulator 1.5.7
```

### Subproject(s) affected?

PartDesign

### Anything else?

FreeCAD returns this:
> Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x76ff9a442520]
#1  0x76ff9c87eead in Data::ElementMap::addName(Data::MappedName&, Data::IndexedName const&, QVector<App::StringIDRef> const&, bool, Data::IndexedName*) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libFreeCADApp.so+0xed
#2  0x76ff9c87fc97 in Data::ElementMap::setElementName(Data::IndexedName const&, Data::MappedName const&, long, QVector<App::StringIDRef> const*, bool) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libFreeCADApp.so+0x297
#3  0x76ff588f91b6 in Sketcher::SketchObject::buildShape() from /tmp/.mount_FreeCA04hdrc/usr/lib/Sketcher.so+0x236
#4  0x76ff588fa1ea in Sketcher::SketchObject::execute() from /tmp/.mount_FreeCA04hdrc/usr/lib/Sketcher.so+0xda
#5  0x76ff9c65d552 in App::DocumentObject::recompute() from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libFreeCADApp.so+0xc2
#6  0x76ff662a23cc in Part::Feature::recompute() from /tmp/.mount_FreeCA04hdrc/usr/lib/./Part.so+0xc
#7  0x76ff9c63773c in App::Document::_recomputeFeature(App::DocumentObject*) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libFreeCADApp.so+0x32c
#8  0x76ff9c643ae5 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libFreeCADApp.so+0x625
#9  0x76ff9c6c21ae in App::DocumentPy::recompute(_object*) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libFreeCADApp.so+0x2ee
#10  0x76ff9c6c22c4 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libFreeCADApp.so+0x24
#11  /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x76ff9ce06068]
#12  /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x76ff9cde5b5c]
#13  /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x76ff9cdf1532]
#14  /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x76ff9ceabd1d]
#15  /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x76ff9ceab794]
#16  /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x76ff9cecabf3]
#17  /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x76ff9cec669a]
#18  /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x76ff9ceba9c9]
#19  0x76ff9c13d48c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libFreeCADBase.so+0x6c
#20  0x76ff9d787924 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libFreeCADGui.so+0x94
#21  0x76ff9d787ab3 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libFreeCADGui.so+0xe3
#22  0x76ff9d787bd4 in Gui::Command::updateActive() from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libFreeCADGui.so+0x44
#23  0x76feb8b57ac7 in SketcherGui::ViewProviderSketch::unsetEdit(int) from /tmp/.mount_FreeCA04hdrc/usr/lib/SketcherGui.so+0x317
#24  0x76ff9db59b73 in Gui::ViewProvider::finishEditing() from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libFreeCADGui.so+0x13
#25  0x76ff9d6fcc6d in Gui::Document::_resetEdit() from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libFreeCADGui.so+0x9d
#26  0x76ff9d67c56a in Gui::Application::setEditDocument(Gui::Document*) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libFreeCADGui.so+0x3a
#27  0x76ff9d711710 in Gui::DocumentPy::resetEdit(_object*) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libFreeCADGui.so+0x30
#28  0x76ff9d711744 in Gui::DocumentPy::staticCallback_resetEdit(_object*, _object*) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libFreeCADGui.so+0x24
#29  /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x76ff9ce06068]
#30  /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x76ff9cde5b5c]
#31  /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x76ff9cdf1532]
#32  /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x76ff9ceabd1d]
[... 20 lines cut ...]
#53  0x76ff9aab6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libQt5Core.so.5+0x108
#54  0x76ff9b9763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#55  /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x76ff9b9c6e6f]
#56  /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x76ff9b9ca0c9]
#57  0x76ff9b9703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#58  0x76ff9d73cc29 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libFreeCADGui.so+0x99
#59  0x76ff9aab6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libQt5Core.so.5+0x108
#60  0x76ff9b1353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libQt5Gui.so.5+0x84c
#61  0x76ff9b10b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libQt5Gui.so.5+0xac
#62  /tmp/.mount_FreeCA04hdrc/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x76ff95213bea]
#63  /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x76ff99110ad7]
#64  /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x76ff99113ac7]
#65  /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x76ff991140ce]
#66  0x76ff9ab0aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libQt5Core.so.5+0x66
#67  0x76ff9aab570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libQt5Core.so.5+0x11b
#68  0x76ff9aabd212 in QCoreApplication::exec() from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libQt5Core.so.5+0x82
#69  0x76ff9d69f362 in Gui::Application::runApplication() from /tmp/.mount_FreeCA04hdrc/usr/bin/../lib/libFreeCADGui.so+0x972
#70  /tmp/.mount_FreeCA04hdrc/usr/bin/freecad(+0x7620) [0x593d4b7d5620]
#71  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x76ff9a429d90]
#72  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x76ff9a429e40]
#73  /tmp/.mount_FreeCA04hdrc/usr/bin/freecad(+0x7981) [0x593d4b7d5981]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-11:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/15154
Please comment your findings there.

closed 2024-07-11 by maxwxyz

oerkel47, 2024-07-11:
Apologies, I don't know how I did not find this.