FreeCAD/FreeCAD #15378: Draft expanded Link Array doesn't obey transformations when executing operations

Anthony-Gaudino, 2024-07-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a Draft Link Array (tested on PointLinkArray), if I expand the array (Expand Array option true) and then transform the individual array items (right click on tree -> transform) then if I boolean the array with a solid the transformations are lost.

Selecting a single element in the array and doing a boolean operation maintains the transformation.

One workaround is to select the array items and create a compound with them, and then use this compound in the boolean operation.

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Launcher
  * Nodes 0.1.36
  * InventorLoader 1.4.0
  * Assembly4.backup1719958896.85335 0.50.12 (Disabled)
  * Render.backup1719959316.120863 2024.6.5 (Disabled)
  * woodworking 0.21.2.33771
  * ThreadProfile.backup1719959316.6397839 1.90.0 (Disabled)
  * MeshRemodel.backup1719959315.871541 1.9.23 (Disabled)
  * Silk 0.1.5
  * Defeaturing 1.2.2
  * Pyramids-and-Polyhedrons
  * Curves.backup1719958901.528519 0.6.36 (Disabled)
  * fasteners.backup1719958901.808955 0.5.20 (Disabled)
  * nurbs
  * sheetmetal.backup1719959316.3711638 0.4.21 (Disabled)
  * A2plus.backup1719958896.309429 0.4.64a (Disabled)
  * Manipulator 1.5.7
  * lattice2 1.0.0
  * CurvedShapes 1.0.9
  * toSketch 1.0.1
  * Curves 0.6.40
  * Assembly4 0.50.15
  * Assembly3 0.12.2
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-11:
Could you try this in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
A demo file would be great to reproduce this.

Anthony-Gaudino, 2024-07-11:
Here's a minimal example file: [link_array_demo.FCStd.zip](https://github.com/user-attachments/files/16182448/link_array_demo.FCStd.zip)

It consists of an sphere with two cubes attached to it's vertexes trough PointLinkArray. One of the cubes has a transformation applied to it.
The boolean operation ignores the transformations.

maxwxyz, 2024-07-12:
Is the bug also present in the weekly builds?

Anthony-Gaudino, 2024-07-12:
> Is the bug also present in the weekly builds?

Yes it does.
I updated the full version info.

Roy-043, 2024-07-24:
The Shape of the modified array is correct as is demonstrated by the compound workaround. But for some reason this shape is not used in the boolean operation. IMO this is Core and not a Draft issue.

LupinIII, 2024-08-08:
I'm adding this here, because I think it is the exact same issue. If it is the bug report title should be changed. If it's not the same issue (someone tell me), I'll open another bug report.

It does not have to be an expanded Draft Link Array for booleans to fail. Any link array with "Link Transform" set to true shows the same behavior. The link array shows the linked objects in the right position (honoring the transform of the original objects), but any operation on the link array will have the result in the wrong position (as if ignoring "Link Transform").

It is not just boolean operations either. Pretty much all operations seem to ignore it. Booleans, "Mirror", "Fillet", even just putting the link array into a compound, will make an object as if the transform wasn't there.
[link_transform_ignored.zip](https://github.com/user-attachments/files/16552975/link_transform_ignored.zip)

Roy-043, 2024-08-09:
@LupinIII IMO that is a different problem. Can you please create a new issue? Thanks.

maxwxyz, 2024-08-11:
@bgbsww any idea here?

bgbsww, 2024-08-11:
No clue.  I take it this isn't right:
[image]

Roy-043, 2024-12-15:
This is still an issue.