FreeCAD/FreeCAD #15379: Segfault on closing Sketch

trohsb, 2024-07-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Open attachment
2. Open Sketch003
3. Click Close button in tasks panel

Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib64/libc.so.6(+0x3c340) [0x7fed19ce4340]
#1  0x7fed1bfeb1eb in Data::ElementMap::addName(Data::MappedName&, Data::IndexedName const&, QVector<App::StringIDRef> const&, bool, Data::IndexedName*) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0xdb
#2  0x7fed1bfebc84 in Data::ElementMap::setElementName(Data::IndexedName const&, Data::MappedName const&, long, QVector<App::StringIDRef> const*, bool) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x274
#3  0x7fec8a452f00 in Sketcher::SketchObject::buildShape() from /usr/lib64/freecad/lib64/Sketcher.so+0x1f0
#4  0x7fec8a4540d4 in Sketcher::SketchObject::execute() from /usr/lib64/freecad/lib64/Sketcher.so+0x2f4
#5  0x7fed1bd9cc8a in App::DocumentObject::recompute() from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x9a
#6  0x7fecb0da5d3f in Part::Feature::recompute() from /usr/lib64/freecad/lib64/Part.so+0xf
#7  0x7fed1bd2120c in App::Document::_recomputeFeature(App::DocumentObject*) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x24c
#8  0x7fed1bd264c1 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x5e1
#9  0x7fed1be090ad in App::DocumentPy::recompute(_object*) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x2ed
#10  0x7fed1be091e7 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x27
#11  /usr/lib64/libpython3.11.so.1.0(+0x114ac5) [0x7fed1b3a8ac5]
#12  /usr/lib64/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x8f) [0x7fed1b365eef]
#13  /usr/lib64/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6916) [0x7fed1b319cc6]
#14  /usr/lib64/libpython3.11.so.1.0(+0x1a6060) [0x7fed1b43a060]
#15  /usr/lib64/libpython3.11.so.1.0(PyEval_EvalCode+0xb1) [0x7fed1b43a121]
#16  /usr/lib64/libpython3.11.so.1.0(+0x1e7783) [0x7fed1b47b783]
#17  /usr/lib64/libpython3.11.so.1.0(+0x1e79a6) [0x7fed1b47b9a6]
#18  /usr/lib64/libpython3.11.so.1.0(PyRun_StringFlags+0x6f) [0x7fed1b47ebbf]
#19  0x7fed1b91ce0c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/lib64/freecad/lib64/libFreeCADBase.so+0x6c
#20  0x7fed1c8df8e7 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x97
#21  0x7fed1c8dfa8c in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0xec
#22  0x7fed1c8dfc96 in Gui::Command::updateActive() from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x46
#23  0x7fec82f69a40 in SketcherGui::ViewProviderSketch::unsetEdit(int) from /usr/lib64/freecad/lib64/SketcherGui.so+0x260
#24  0x7fed1ccfc4a7 in Gui::ViewProvider::finishEditing() from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x17
#25  0x7fed1c82dcd9 in Gui::Document::_resetEdit() from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x99
#26  0x7fed1c7ad051 in Gui::Application::setEditDocument(Gui::Document*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x41
#27  0x7fed1c85a2c1 in Gui::DocumentPy::resetEdit(_object*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x31
#28  0x7fed1c85a2f7 in Gui::DocumentPy::staticCallback_resetEdit(_object*, _object*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x27
#29  /usr/lib64/libpython3.11.so.1.0(+0x114ac5) [0x7fed1b3a8ac5]
#30  /usr/lib64/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x8f) [0x7fed1b365eef]
#31  /usr/lib64/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6916) [0x7fed1b319cc6]
#32  /usr/lib64/libpython3.11.so.1.0(+0x1a6060) [0x7fed1b43a060]
#33  /usr/lib64/libpython3.11.so.1.0(PyEval_EvalCode+0xb1) [0x7fed1b43a121]
#34  /usr/lib64/libpython3.11.so.1.0(+0x1e7783) [0x7fed1b47b783]
#35  /usr/lib64/libpython3.11.so.1.0(+0x1e79a6) [0x7fed1b47b9a6]
#36  /usr/lib64/libpython3.11.so.1.0(PyRun_StringFlags+0x6f) [0x7fed1b47ebbf]
#37  0x7fed1b91ce0c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/lib64/freecad/lib64/libFreeCADBase.so+0x6c
#38  0x7fed1c8df8e7 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x97
#39  0x7fed1c8dfa8c in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0xec
#40  0x7fec82fbb27f in SketcherGui::TaskDlgEditSketch::reject() from /usr/lib64/freecad/lib64/SketcherGui.so+0x13f
#41  0x7fed1cc605c8 in Gui::TaskView::TaskView::reject() from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x68
#42  /usr/lib64/libQt5Core.so.5(+0x2acc22) [0x7fed1a3a7c22]
#43  /usr/lib64/libQt5Widgets.so.5(+0x2a4af8) [0x7fed1aec0af8]
#44  /usr/lib64/libQt5Core.so.5(+0x2acd30) [0x7fed1a3a7d30]
#45  0x7fed1ae13ff6 in QAbstractButton::clicked(bool) from /usr/lib64/libQt5Widgets.so.5+0x46
#46  /usr/lib64/libQt5Widgets.so.5(+0x1f826e) [0x7fed1ae1426e]
#47  /usr/lib64/libQt5Widgets.so.5(+0x1f9e7b) [0x7fed1ae15e7b]
#48  0x7fed1ae160a7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib64/libQt5Widgets.so.5+0x107
#49  0x7fed1ad61a78 in QWidget::event(QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x268
#50  0x7fed1ad1ed32 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x82
#51  0x7fed1ad272af in QApplication::notify(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x12df
#52  0x7fed1c891a18 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x98
#53  0x7fed1a374d08 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib64/libQt5Core.so.5+0x138
#54  0x7fed1ad25459 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib64/libQt5Widgets.so.5+0x1c9
#55  /usr/lib64/libQt5Widgets.so.5(+0x15ebd7) [0x7fed1ad7abd7]
#56  /usr/lib64/libQt5Widgets.so.5(+0x161f4f) [0x7fed1ad7df4f]
#57  0x7fed1ad1ed32 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib64/libQt5Widgets.so.5+0x82
#58  0x7fed1c891a18 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x98
#59  0x7fed1a374d08 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib64/libQt5Core.so.5+0x138
#60  0x7fed1a75929f in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib64/libQt5Gui.so.5+0x92f
#61  0x7fed1a737b9b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt5Gui.so.5+0xab
#62  /usr/lib64/libQt5XcbQpa.so.5(+0x65f7e) [0x7fed14bcef7e]
#63  /usr/lib64/libglib-2.0.so.0(+0x57834) [0x7fed18285834]
#64  /usr/lib64/libglib-2.0.so.0(+0x5a977) [0x7fed18288977]
#65  /usr/lib64/libglib-2.0.so.0(g_main_context_iteration+0x30) [0x7fed18288fa0]
#66  0x7fed1a3c65ba in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt5Core.so.5+0x6a
#67  0x7fed1a37372b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt5Core.so.5+0x12b
#68  0x7fed1a37b924 in QCoreApplication::exec() from /usr/lib64/libQt5Core.so.5+0x94
#69  0x7fed1c7ba5f1 in Gui::Application::runApplication() from /usr/lib64/freecad/lib64/libFreeCADGui.so+0xbe1
#70  /usr/lib64/freecad/bin/FreeCAD(+0x9267) [0x558a3242e267]
#71  /usr/lib64/libc.so.6(+0x262e0) [0x7fed19cce2e0]
#72  /usr/lib64/libc.so.6(__libc_start_main+0x89) [0x7fed19cce399]
#73  /usr/lib64/freecad/bin/FreeCAD(+0x95b5) [0x558a3242e5b5]

### Full version info

```shell
OS: Gentoo Linux (KDE)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.Unknown
Build type: RelWithDebInfo
Python 3.11.9, Qt 5.15.14, Coin 4.0.2, Vtk 9.2.5, OCC 7.7.0
Locale: English/United Kingdom (en_GB)
```

### Subproject(s) affected?

PartDesign

### Anything else?

[Snowman.zip](https://github.com/user-attachments/files/16182154/Snowman.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-07-11:
Another duplicate of #15102! Nyaha!

maxwxyz, 2024-07-11:
Duplicate of #15102

closed 2024-07-11 by maxwxyz