FreeCAD/FreeCAD #15380: [TNP] TNP fails where V21 works.

shaise, 2024-07-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm working with the latest builds of FreeCAD. 
It looks like I have many cases where changes break the model.
I have no problem if the changes are so big that even TNP mitigation can't handle it, But my problem is that even simple changes that do not change the model structure at all - still breaks. I'm talking about things that works with the not TNP mitigated V21, but does not work now. 
In The attached example I have 3 files, one for the latest build, one for version 21, and one for LS3. This file works both on v0.21 and LS but brakes on dev build.

How to reproduce:
1. open the file
2. open 'Sketch', under the top 'Pad'
3. change bottom dimension from 2.2mm to 2.4mm
Model breaks (fillets loose original place)

[tnpTests.zip](https://github.com/user-attachments/files/16182379/tnpTests.zip)

One interesting observation: If I open the file made in v0.21 with the dev version and repeat the procedure, this time the model does not break.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git)
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * BIM_Workbench 2021.12.0
  * fasteners 0.5.24
  * Manipulator 1.5.7
  * OpenTheme 2024.7.3
  * PieMenu 1.7.4
  * Render 2024.6.5
  * SelectorToolbar
  * sheetmetal 0.4.22
  * TabBar (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bgbsww, 2024-07-11:
Thanks for the example!
In the broken case (FilterPunchTest) we get:

15:52:52  <PropertyLinks> PropertyLinks.cpp(452): FilterPunchTest#Sketch001.AttachmentSupport missing element reference FilterPunchTest#Pad ;g3;SKT;:Hdac,E;FAC;:Hdac:4,F;:G0;XTR;:Hdac:8,F.Face15
15:52:52  <PropertyLinks> PropertyLinks.cpp(452): FilterPunchTest#Sketch001.ExternalGeometry missing element reference FilterPunchTest#Pad ;g8;SKT;:Hdac,E;:G;XTR;:Hdac:7,F;:U;XTR;:Hdac:7,E.Edge31
15:52:52  <PropertyLinks> PropertyLinks.cpp(452): FilterPunchTest#Sketch001.ExternalGeometry missing element reference FilterPunchTest#Pad ;g6;SKT;:Hdac,E;:G;XTR;:Hdac:7,F;:U;XTR;:Hdac:7,E.Edge37
15:52:52  <PropertyLinks> PropertyLinks.cpp(452): FilterPunchTest#Sketch001.ExternalGeometry missing element reference FilterPunchTest#Pad ;g4;SKT;:Hdac,E;:G;XTR;:Hdac:7,F;:U;XTR;:Hdac:7,E.Edge4
15:52:52  Invalid shape name ?Edge31
Clear constraints to external geometry
15:52:52  <PropertyLinks> PropertyLinks.cpp(452): FilterPunchTest#Fillet.Base missing element reference FilterPunchTest#Pad001 ;g9v1;SKT;:Hdac,V;:G;XTR;:Hdac:7,E;:Hdc6,E.Edge57
15:52:52  <PropertyLinks> PropertyLinks.cpp(432): FilterPunchTest#Fillet.Base auto change element reference FilterPunchTest#Pad001 ;g12v1;SKT;:Hdac,V;:G;XTR;:Hdac:7,E;:Hdc6,E.Edge51 -> ;g25v1;SKT;:Hdac,V;:G;XTR;:Hdac:7,E;:Hdc6,E.Edge51
15:52:52  <PropertyLinks> PropertyLinks.cpp(432): FilterPunchTest#Fillet.Base auto change element reference FilterPunchTest#Pad001 ;g13v1;SKT;:Hdac,V;:G;XTR;:Hdac:7,E;:Hdc6,E.Edge49 -> ;g26v1;SKT;:Hdac,V;:G;XTR;:Hdac:7,E;:Hdc6,E.Edge49
15:52:52  <PropertyLinks> PropertyLinks.cpp(432): FilterPunchTest#Fillet.Base auto change element reference FilterPunchTest#Pad001 ;g3v2;SKT;:Hdac,V;:G;XTR;:Hdac:7,E;:Hdc6,E.Edge47 -> ;g16v2;SKT;:Hdac,V;:G;XTR;:Hdac:7,E;:Hdc6,E.Edge47
15:52:52  <PropertyLinks> PropertyLinks.cpp(432): FilterPunchTest#Fillet.Base auto change element reference FilterPunchTest#Pad001 ;g3v1;SKT;:Hdac,V;:G;XTR;:Hdac:7,E;:Hdc6,E.Edge45 -> ;g16v1;SKT;:Hdac,V;:G;XTR;:Hdac:7,E;:Hdc6,E.Edge45
15:52:52  <PropertyLinks> PropertyLinks.cpp(432): FilterPunchTest#Fillet.Base auto change element reference FilterPunchTest#Pad001 ;g14v1;SKT;:Hdac,V;:G;XTR;:Hdac:7,E;:Hdc6,E.Edge43 -> ;g27v1;SKT;:Hdac,V;:G;XTR;:Hdac:7,E;:Hdc6,E.Edge43
15:52:52  <PropertyLinks> PropertyLinks.cpp(452): FilterPunchTest#Fillet.Base missing element reference FilterPunchTest#Pad001 ;g6v1;SKT;:Hdac,V;:G;XTR;:Hdac:7,E;:Hdc6,E.Edge37
15:52:52  <Exception> FeatureDressUp.cpp(209): std::vector<Part::TopoShape> PartDesign::DressUp::getContinuousEdges(const PartDesign::TopoShape&) -- Invalid edge link: ;g9v1;SKT;:Hdac,V;:G;XTR;:Hdac:7,E;:Hdc6,E.Edge57
15:52:52  Fillet: Invalid edge link: ;g9v1;SKT;:Hdac,V;:G;XTR;:Hdac:7,E;:Hdc6,E.Edge57
15:52:52  <Exception> FeatureDressUp.cpp(209): std::vector<Part::TopoShape> PartDesign::DressUp::getContinuousEdges(const PartDesign::TopoShape&) -- Invalid edge link: ;g9v1;SKT;:Hdac,V;:G;XTR;:Hdac:7,E;:Hdc6,E.Edge57
15:52:52  Fillet: Invalid edge link: ;g9v1;SKT;:Hdac,V;:G;XTR;:Hdac:7,E;:Hdc6,E.Edge57

In the FilterPunchTest21 file, even if you recompute, there is no lookup failure warning:

15:54:33  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: FilterPunchTest21#Sketch
15:54:33  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: FilterPunchTest21#Body
15:54:33  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: FilterPunchTest21#Pad
15:54:33  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: FilterPunchTest21#Pad
15:54:33  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: FilterPunchTest21#Sketch001
15:54:33  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: FilterPunchTest21#Pad001
15:54:33  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: FilterPunchTest21#Pad001
15:54:33  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: FilterPunchTest21#Fillet
15:54:33  <App> PropertyTopoShape.cpp(425): Pending recompute for generating element map: FilterPunchTest21#Fillet
15:54:47  Sketcher External Geometry without ExternalGeometryExtension: 78e43ee4-1f2d-4a8c-8fed-e1d4fcc06b84 
15:54:47  Sketcher External Geometry without ExternalGeometryExtension: dc865345-b4ef-49b8-b1a3-e6359b5415a3 
15:55:34  Sketcher External Geometry without ExternalGeometryExtension: ec7f68ec-5ded-4051-88db-0bad179db805 
15:55:34  Sketcher External Geometry without ExternalGeometryExtension: c79ac513-7bd1-4f06-955f-264cf6bd2432 

Looking at the new ElementMap for the Pad, we can see this mismatch for Face15:
;g3;SKT;:Hdac,E;FAC;:Hdac:4,F;:G0;XTR;:Hdac:8,F
g39;SKT;:Hdac,E;FAC;:Hdac:4,F;:G0;XTR;:Hdac:8,F

Files from pre .22 will not have element maps, so nothing will be loaded.  Files from LS3 will have element maps that cannot be read by .22 so nothing will be loaded.

App.ActiveDocument.Pad.Shape.ElementReverseMap  shows the presence or absence of the elementMaps - in V21 file case, you can see that it is empty on initial read, but gets populated either with a recompute or the sketch edit.

If you save the v21 file again after creating an elementmap, with or without the sketch change, you end up with a file that then does fail when you reopen it and edit the sketch.

That tends to support the hypothesis that there is a load / save problem with elementmaps - files opened without them that generate new ones work, files that read them in fail.

The elementmaps are supposed to get stripped / sanitized when written out, and repopulated when read back in due to potential memory address or other variations, especially across platforms.  This seems not to be working correctly.

shaise, 2024-07-12:
@bgbsww ,
Once again, thank you so much for your epic work on this issue!

shai

shaise, 2024-07-12:
@bgbsww ,
If it helps, I recreated the file from scratch, and indeed without saving, everything works!
I do get the following warnings if it is relevant:
```
09:27:13  Sketcher External Geometry without ExternalGeometryExtension: 8afedd85-4aba-455d-8914-57c500648d76 
09:27:13  Sketcher External Geometry without ExternalGeometryExtension: a03bab5f-04d1-4049-8339-57c9ba34189a 
```

bgbsww, 2024-07-23:
With the latest, not quite yet merged, but fixes save/load,  I can open the good FilterPucnhTest21.FCStd file, recompute it, change the constraints, save it, close it, reopen it, change the constraints, seems to be fixed.

shaise, 2024-07-23:
Great!

maxwxyz, 2024-07-29:
This should be fixed with #15575 
Please test again with the latest weekly tomorrow: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

FEA-eng, 2024-07-30:
This one indeed seems to be fixed but maybe @shaise could recheck too just in case.

shaise, 2024-07-30:
Testing with the files I attached the model still breaks, but, I guess it is because the file was not saved correctly in the first place.
After fixing the model an re-saving it, everything works fine. 
More important is that loading V21 files works as well. - so this is very good.
Thank you @bgbsww for all the work! closing this issue.