FreeCAD/FreeCAD #15386: Assembly: BOM shortcut (V) clashes with multiple view shortcuts

furgo16, 2024-07-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the Assembly workbench, the BOM creation keyboard shortcut (<kbd>V</kbd>) clashes with multiple other global shortcuts:

- View Fit All (<kbd>V</kbd>, <kbd>F</kbd>)
- View Fit Selection (<kbd>V</kbd>, <kbd>S</kbd>)
- Drawing mode: From "As is"  (<kbd>V</kbd>, <kbd>1</kbd>) to "Flat lines"  (<kbd>V</kbd>, <kbd>2</kbd>)

Since the above are so ubiquitous and global, it might be best to choose a different keyboard shortcut for BOM creation.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38120 (Git) Snap 1073
Build type: Release
Branch: main
Hash: 9eeca1f6ad50cea9fa2d30d1a4fbb8b6e8fb4bbd
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.7.3
  * Curves 0.6.40
  * ezydxf
  * Ondsel-Lens 2024.7.5.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-07-18:
Sure. I just took one letter I thought available. I'll asign another one.

PaddleStroke, 2024-07-18:
It's not the BOM tool that has V shortcut it's exploded views.

closed 2024-07-22 by chennes