FreeCAD/FreeCAD #15387: Assembly: BOM conflates sub-assemblies into one if they have the same name

furgo16, 2024-07-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

## Steps to reproduce

1. Uncompress the [Cubecyl.zip](https://github.com/user-attachments/files/16192054/Cubecyl.zip) file
2. Open the `CubecylAssembly.FCStd`
3. Double-click on the Assembly
4. Click on the Create BOM button
5. Tick the Sub-assemblies children checkbox
6. Click OK
7. **Expected**: the BOM contains 4 different components
8. **Actual**: the BOM contains only 2 different components. Both sub-assemblies are assumed to be the same and are conflated into one entry in the BOM.

[image]

[image]

This seems to be because the external files all have an assembly named `Assembly` (same name). Perhaps also because they have the same number of components, named the same (I'm not sure how deep this is checked).

Renaming one of the assemblies in one of the external files (e.g. `AssemblyOther`) works around this and then displays the expected number of components.

[image]
  

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38120 (Git) Snap 1073
Build type: Release
Branch: main
Hash: 9eeca1f6ad50cea9fa2d30d1a4fbb8b6e8fb4bbd
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.7.3
  * Curves 0.6.40
  * ezydxf
  * Ondsel-Lens 2024.7.5.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-12:
Not sure how the app other than the name would know if the sub parts are the same or not...

furgo16, 2024-07-13:
I cannot really answer the question. The only thing that comes to mind is that they come from two separate files.

tobiasfalk, 2024-07-15:
Maybe with a unique hash that is created  from the file?

PaddleStroke, 2024-07-18:
Currently we are only comparing names. To solve this we need to also compare the document name.
Perhaps the name column should even use the doc#obj names?
So the name column would show : 
CubeCyl1#Assembly
CubeCyl1#Cylinder
CubeCyl1#Cube
CubeCyl2#Assembly
CubeCyl2#Cylinder
CubeCyl2#Cube

Does that sound better ?

PaddleStroke, 2024-07-18:
Or there can also be a doc name column.

closed 2024-07-22 by chennes