FreeCAD/FreeCAD #15388: Techdraw: Balloon annotation pointer does not track reference in moves

johnsonm, 2024-07-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In TechDraw, create a balloon annotation, then drag the created number. The tip of the arrow should keep pointing at the same spot in the drawing that was selected originally. Instead, the entire arrow moves with the balloon.

Rename [BrokenBalloon.zip](https://github.com/user-attachments/files/16195229/BrokenBalloon.zip) to BrokenBalloon.FCStd, go to the Page, grab the **1**, and drag it. The arrow point will not stay attached to the cube vertex; instead, the arrow keeps its shape and the point of the arrow will point at nothing.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38126 (Git)
Build type: Unknown
Branch: main
Hash: ac2b65c37b18617101822e5ece06f3c9b21375cd
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.5.24
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
  * Assembly3 0.12.2
  * DynamicData 2.62.0
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-07-12:
Forum discussion: https://forum.freecad.org/viewtopic.php?t=89061

WandererFan, 2024-07-12:
This is due to a conflict between multi-selection and moving the entire balloon.  Both use the ctrl key.  

If you turn off multi-select, you should be able to drag the balloon while leaving the arrow position unchanged.

johnsonm, 2024-07-12:
I wasn't using the control key though...

johnsonm, 2024-07-12:
Wow that's an interesting layering violation! Does Qt make this hard somehow?

Seems like sketcher handles this differently?

johnsonm, 2024-07-13:
I hadn't even known that control-drag would move the whole arrow before. I had been laboriously moving the tip by editing its X and Y coordinates! ☺

WandererFan, 2024-07-13:
<<Wow that's an interesting layering violation! Does Qt make this hard somehow?>>

The only way to select multiple items in a QGraphicsScene is to use the ctrl key, so we have to fake it behind the scenes.   

TD and the dependency graph are the only users of QGraphicsScene in FC.

WandererFan, 2024-07-17:
The easy fix here is to change the Balloon drag modifier to something that doesn't conflict with anything else.  Shift+Control seems available, at least on linux. 

Anybody know if Shift+Control has a special meaning on Mac or Win?

johnsonm, 2024-07-17:
Hmm. Since multi-selection is off-by-default I'd hate for those of us who want it to create a tax on folks who don't. Or would shift+control apply only in multi-select mode?

WandererFan, 2024-09-16:
solution proposal:

If multiselect is turned off, balloon dragging is unchanged.  Drag to move the bubble, ctrl+drag to move the bubble and the arrow.

If multiselect is turned on,
    if multiselect.modifier == balloon.origindrag.modifer
        this is the default configuration using the Ctrl key for both.
        we will now treat all drags as bubble drags in this case
        
    if muliselect.modifier != balloon.origindrag.modifier
        this is the configuration if a different key has been assigned as balloon.origindrag.modifier
        treat drags according to the press/no press of balloon.origindrag.modifier

There is a new preference on the advanced tab - balloon drag override modifiers.  Controls use of default (ctrl)
modifier or some other combination of Shift/Alt/Ctrl/Meta keys to signal that drag should move bubble 
and arrow together.  This setting will depend on OS and local key mappings.  An alternate modifier 
combination may work on linux, but not on Win.

WandererFan, 2024-09-18:
draft PR for proposed solution:  https://github.com/FreeCAD/FreeCAD/pull/16639

commit 31220628 (2024-09-30): [TD]allow override of Balloon drag modifier (fix #15388)
commit e41def8c (2024-09-30): Merge pull request #16639 from WandererFan/balloonDrag15388

[TD]Balloon origin drag modifiers (fix #15388)
commit 3bccd5bd (2024-09-30): [TD]allow override of Balloon drag modifier (fix #15388)
closed 2024-09-30 by yorikvanhavre