FreeCAD/FreeCAD #15389: Unable to copy objects in sketch

ShamanTcler, 2024-07-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Trying to copy an ellipse in sketcher ... in an attempt to "get something" I input 3 for both the copies and rows inputs, I get nothing.

And "Close" just doesn't feel like the right name. Is it also missing a "Cancel" button?
 
[image]

The issue is in the file Back.FCStd, but included all file in the set as the sketch in Back refers to a spreadsheet in CraftsmanCoatRack.FCStd

### Full version info

```shell
OS: Windows 11 build 22635
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git)
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.65
  * btl 0.9.9
  * Curves 0.6.40
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Anything else?

[CraftmanCoatRack.zip](https://github.com/user-attachments/files/16195551/CraftmanCoatRack.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-12:
Please describe how to reproduce this, I think you do not use it correctly: https://wiki.freecad.org/Sketcher_Translate
You have to use the 3D view to define the points.

closed 2024-07-12 by maxwxyz

ShamanTcler, 2024-07-12:
Agreed, I did not use it properly.... after reading the link and playing a bit I did get it to work... it was not obvious at least to me that what was required was mouse selections in the drawing window. (I am pretty certain this has caught me before.)

Once the translate/copy is performed is there a way to drive it parametrically as with other sketcher elements?

The use case is:
Spreadsheet determines the offset from an edge of element 1 then makes 1:N more copies of the element at a given X/Y offset. Think coat hooks on a variable length coat rack