FreeCAD/FreeCAD #15390: Involute gear in V21.2 under Part Design workbench

Dan012954, 2024-07-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While utilizing the Involute Gear option, it seems that in the Involute Parameters, the parameter shown as Module needs to be entered as the value of the gear Addendum (in inches) for the gear to be properly constructed with the appropriate diameters (pitch diameter (in), outside diameter (in)).

### Full version info

```shell
See "Anything else" comments below.
```

### Subproject(s) affected?

PartDesign

### Anything else?

Using widely published gear parameter relationships and other means of gear design (outside of FreeCAD), a gear with 8 teeth, pitch diameter 1.14(in) and diametral pitch of 7 produces an addendum of .143(in). This gear cannot be created in FreeCAD using these parameters or the metric equivalents (Module, mm). It can only be created fairly close to size by entering the addendum (in inches) for the Module entry. So... It is useable but is not straight forward to use without knowledge of this anomaly.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-12:
Is this bug also present in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Which default unit system do you use?

Dan012954, 2024-07-12:
No idea regarding dev version.I'm using inch units.Sent from my Galaxy

luzpaz, 2024-07-24:
@Dan012954 we need your full [About](https://wiki.freecad.org/About) info version.

luzpaz, 2024-07-24:
Hi @looooo would you have a moment to troubleshoot this ?

maxwxyz, 2024-08-05:
Any news on this? Otherwise I'll close this issue.

Dan012954, 2024-08-05:
Max,As prompted in a previous email, attached please find a PDF containing my complete "About" information, (see below).Best regardsDan Andrews

maxwxyz, 2024-08-06:
There are no attachments. Please update the post on GitHub.

Dan012954, 2024-08-06:
I have never used Git Hub.Here is another try at attachment.(I checked and it was included in previous mail!?)

maxwxyz, 2024-08-13:
I'm closing this. We need full about info. Ideally is also an example file or macro of how the model is created and if this issue is still existing in the current dev versions. Feel free to reopen with the requested info if there is still an issue.

closed 2024-08-13 by maxwxyz